armink/EasyFlash #4: 移植出现的部分问题

JGINGIT, 2015-08-27:
- ```flash_env``` 在```easyflash\port\ef_port.c``` Line32 出现，但是源码中并没有```flash_env```的定义，参考demo得知应该为```ef_env```。
- ```easyflash\port\ef_port.c```中应该增加```#include <stdarg.h>```防止开了```PRINT_DEBUG``` 编译出错。

armink, 2015-08-27:
之前做过一次重构，所以部分细节有些遗漏，多谢反馈哈~

armink, 2015-08-27:
@JGINGIT 现在已经移植成功了吗？

JGINGIT, 2015-08-27:
@armink 还没，代码还没仔细研究，理解不了```EF_ERASE_MIN_SIZE```和```ENV_USER_SETTING_SIZE```。例如我是W25Q16最小擦除单位是4KB，所以```EF_ERASE_MIN_SIZE = （4 * 1024) ```？

```ENV_USER_SETTING_SIZE ```是用户设置的数据环境变量的空间大小么？
``` C
    /* check capacity of ENV  */
    if (env_str_len + get_env_data_size() >= ENV_USER_SETTING_SIZE) {
        return EF_ENV_FULL;
    }
```
但是又为什么有```ENV_AREA_SIZE```的定义，用户数据环境变量空间大小又等于``` ENV_AREA_SIZE - ENV_SYSTEM_SIZE```?

armink, 2015-08-27:
@JGINGIT 
- 1、`W25Q16`的`EF_ERASE_MIN_SIZE = （4 * 1024)` 这个是没问题的
- 2、对于`ENV_AREA_SIZE`与`ENV_USER_SETTING_SIZE`的区别，在[这里](https://github.com/armink/EasyFlash/blob/master/docs/zh/port.md#55-备份区)有解释

> - 2、由于EasyFlash对所有的环境变量都会使用RAM缓存，但是在更多时候用户使用的环境变量大小会比EF_ERASE_MIN_SIZE小，所以需要再定义ENV_USER_SETTING_SIZE来指定用户设定的环境变量大小。 
- 3、环境变量区总容量在不同的模式下会有差异
  - 1、常规模式：没有差异；
  - 2、擦写平衡模式：系统区将会占用1个EF_ERASE_MIN_SIZE大小，数据区至少等使用2个以上Flash扇区；
  - 3、掉电保护模式：环境变量区将会被备份，所以总容量是常规模式的2倍；
  - 4、擦写平衡+掉电保护模式：系统区将会占用1个EF_ERASE_MIN_SIZE大小，数据区将会是擦写平衡模式下的数据区总容量的2倍。

所以，你先看下这里的解释能不能理解？

closed 2015-09-11 by armink