armink/EasyFlash #19: Updating flash when new variables are created

crespum, 2018-05-23:
I've idefntified an scenario that is quite common in my opinion but it's not supported by the library (and I'm not sure if it should).

Imagine you are using this set of variables:

```c
/* ef_ports.c default environment variables set for user */
static const ef_env default_env_set[] = {
    {"var1", "foo"},
};
```

But in the next release of your code you want to add a new one (or even remove one of the existing ones). For instance you might need this:

```c
/* ef_ports.c default environment variables set for user */
static const ef_env default_env_set[] = {
    {"var1", "foo"},
    {"var2", "6"}
};
```

Since you already have some variables saved in memory, your code will not be able to find `"var3"` unless you call manually call `ef_env_set_default()` (the library will not do it). To call it only when a new variable is added I've done this:

```c
/* ef_ports.c default environment variables set for user */
static const ef_env default_env_set[] = {
    {"__ENV_VER", "1"}, // Increase this number whenever a new variable is added 
    {"var1", "foo"},
    {"var2", "6"}
};
```
```c
/* main.c */
easyflash_init();

char *var;
var = ef_get_env("__ENV_VER");
if (var == NULL || memcmp(var, EF_ENV_ID, sizeof(EF_ENV_ID))) {
    ef_env_set_default();
}
```
---
**My question is, should this be implemented by the library somehow or should be part of the application layer? Is there any other way around?**

armink, 2018-05-23:
Auto set_default was implemented. But I have a problem. 

If the kv `{"var1", "foo"}` is changed to **`{"var1", "bar"}` . When updating ENV, do you want to change the `var1` 's new value (bar) to default (foo)?

crespum, 2018-05-24:
Any option is valid in my case. In general I would accept any of them as long as you state it clearly in the docs.

armink, 2018-05-24:
The new feature is added, please see: https://github.com/armink/EasyFlash/tree/3.1.0

Please defined `EF_ENV_AUTO_UPDATE` and `EF_ENV_VER_NUM` on you `ef_cfg.h`. Then new ENV on default set will be added to system when `EF_ENV_VER_NUM` is changed. You can test it on your board.

Thx for your feedback :)

crespum, 2018-05-30:
Closing the issue as it's been implemented in https://github.com/armink/EasyFlash/commit/d199a4d3eb4a35c88b73bd55cf04a24f8bf5d72e. Nice work @armink!

closed 2018-05-30 by crespum