armink/EasyFlash #20: 平衡磨损配置疑问

chenxiuqimr, 2018-06-20:
根据easyflash文档说明有以下一段话
       例如：EF_ERASE_MIN_SIZE是128K，ENV_USER_SETTING_SIZE是2K， 那么你可以这样定义不同模式下的环境变量总容量：
      擦写平衡模式：3*EF_ERASE_MIN_SIZE（它将会有3个Flash扇区去存储环境变量， 1个系统区,2个数据区，按照每个Flash扇区可被擦写10W次计算，那么当前配置至少可擦写20W次）;

那么假如 EF_ERASE_MIN_SIZE是4K，ENV_USER_SETTING_SIZE是2K，flash扇区寿命为10k次， 擦写平衡模式下我要如何配置才能支持可擦写1000k次呢？

见到demo平衡模式下，ENV_AREA_SIZE都是配置为3*EF_ERASE_MIN_SIZE大小，该模式下我如果设置ENV_AREA_SIZE为30*EF_ERASE_MIN_SIZE会有什么后果呢，对flash的使用寿命会有什么影响呢？

armink, 2018-06-21:
> 那么假如 EF_ERASE_MIN_SIZE是4K，ENV_USER_SETTING_SIZE是2K，flash扇区寿命为10k次， 擦写平衡模式下我要如何配置才能支持可擦写1000k次呢？

因为 1000k= 10*10W，所以定义 ENV_AREA_SIZE 为 10 倍的 EF_ERASE_MIN_SIZE 就行了

armink, 2018-06-21:
目前是这样的策略

chenxiuqimr, 2018-06-21:
1 每次执行ef_save_env会轮流擦写到不同的扇区。
2 每次执行ef_save_env都会先尝试操作第一个扇区，写坏后第一个扇区后才开始写第二个扇区
老大，用的是方法1还是方法2呢？

armink, 2018-06-21:
目前是第二种

chenxiuqimr, 2018-06-21:
明白，谢谢！easyflash很棒

closed 2018-06-21 by chenxiuqimr