armink/EasyFlash #25: 自动更新问题

outely, 2018-08-08:
在开启**EF_ENV_AUTO_UPDATE**后，每次重新开机都会擦写Flash。
是否是缺少在`ef_env_init()`-->`result = ef_load_env()`中加载Flash中的**EF_ENV_VER_NUM**到`env_cache[ENV_PARAM_INDEX_VER_NUM]`？
还是我配置或使用方式不对:smiley:

armink, 2018-08-08:
> 在开启EF_ENV_AUTO_UPDATE后，每次重新开机都会擦写Flash。
正常是只有你修改了EF_ENV_VER_NUM 才会更新 env。

怀疑你是不是用的 demo 中的那个 boot counts 功能，那个是每次开机都自动加一，然后存储的。

outely, 2018-08-08:
我不是使用demo.
我尝试在`EfErrCode ef_load_env(void)`中增加
`ef_port_read(get_env_system_addr() + ENV_PARAM_INDEX_VER_NUM * 4,
&env_cache[ENV_PARAM_INDEX_VER_NUM] , 4);`后，就不会出现每次重新开机都会擦写Flash了。因为我找不到其它地方有开机加载Flash中的EF_ENV_VER_NUM到env_cache[ENV_PARAM_INDEX_VER_NUM]。

armink, 2018-08-08:
好像确实有问题。方便提一个 pr 吗？

armink, 2018-08-08:
在 env_auto_update 里加感觉就可以了

outely, 2018-08-08:
嗯，在`env_auto_update()`中添加更方便。😃