armink/EasyFlash #30: ENV_USER_SETTING_SIZE 是不是只能小于一个section？

e190, 2018-10-10:
假如我想把一个flash的大部分空间都拿来 做 KV数据库用，怎么办？

armink, 2018-10-10:
目前好像不能这样做，除非有足够多的 RAM 。后面有规划一种全新模式，会去掉 RAM 缓存

e190, 2018-10-10:
我想的是，如果最小擦除是4K，那RAM缓存就设置为4K。然后每次写之前都要先读出来，然后擦除，然后写入。我这种办法是不是很笨的？

armink, 2018-10-10:
> 我想的是，如果最小擦除是4K，那RAM缓存就设置为4K

不是你说的这样的，ENV FLASH 区域大小、最小擦除粒度以及 RAM 缓存都是可以独立设置的。

qinhao4832, 2018-10-16:
[Flash](..\Src\easyflash\src\ef_env.c:152) ENV start address is 0x00000000, size is 4096 bytes.
[Flash](..\Src\easyflash\src\ef_env.c:821) Calculate ENV CRC32 number is 0xA28EF931.
[Flash](..\Src\easyflash\src\ef_env.c:833) Verify ENV CRC32 result is OK.
[Flash]EasyFlash V3.2.3 is initialize success.
[Flash]You can get the latest version on https://github.com/armink/EasyFlash .
iap_need_copy_app=0
iap_copy_app_size=0
stop_in_bootloader=0
device_id=1
jsonARM198708030015={"cmd":101,"s_uis_id":255,"s_uis_type":255,"fn":0,"result":1,"end":101}

boot_times=17
json_198708030015={"cmd":101,"s_uis_id":255,"s_uis_type":255,"fn":0,"result":1,"end":101}

mode: normal
size: 288/2048 bytes.
The system now boot 18 times

以上是我的的env输出，键值 是 json 字符串。 
我也想要要存储几百条这样的json字符串，需要一个几十K大小的KV数据库，但STM32F103芯片 没有那么ram可以缓存。 
请问，类似我这样的应用，该怎么使用easyflash？

armink, 2018-10-16:
@qinhao4832 如果是这种情况，当前确实没有比较适合你的模式。等 EasyFlash 的零 RAM 占用模式开发完成后，就可以用了。

armink, 2019-08-23:
最新发布版本已经不在需要 RAM 缓存了，极大的降低了资源占用，可以试试

closed 2019-08-23 by armink