armink/EasyFlash #43: 询问保存耗时过长解决问题

shuhuahuang, 2019-03-17:
一直想找SPI FLASH的磨损均衡方案，直到发现此项目，感觉很符合自己的需求，非常感谢开源。在使用中遇到问题请教一下：
1，在测试保存数据时，当扇区使用差不多的时候，在此时保存数据，程序会释放其它扇区数据，导致保存数据耗时过长，而对无操作系统来说，可能会导致看门狗复位，建议后续增加喂狗接口；
2，另外有10个扇区，比如从1-10，当用到5时（1-5都dirty），当此时把1-4都清理，那当5用完时，会从1开始用还是从6开始到10，而从1再开始？
3、若是从6开始，能否开放清理扇区的接口，在程序空闲时调用，这样在1的时候，保存数据，会耗时降低一些。期待你的回答，再次感谢开源，谢谢。

armink, 2019-03-20:
- 1、喂狗与 EasyFlash 没有强依赖的业务关联，所以不太可能增加对应接口。我建议可以改进下你的 EasyFlash 对接的 Flash 擦除接口，擦除时间过长时，增加喂狗等操作。
- 2、默认 GC 的扇区阈值是 1 ，所以到了第五个扇区，即便前面都是脏扇区也不会去 GC ，只有到了第十个才会执行 GC。

shuhuahuang, 2019-03-21:
未带系统的程序，保存一个数据，恰好碰到GC，这样会导致保存耗时很久，其中牵扯到移动数据，擦除扇区操作，而这一直卡在easyflash里面吧，后续的程序得不到执行吧

armink, 2019-03-21:
所以，你需求是想定期的执行 GC 对吗？

shuhuahuang, 2019-03-22:
嗯  我觉得实际使用中加入 闲时来检测是否需要整理扇区数据，并且每次整理扇区的耗时不要过长，比如整理一块扇区，然后跳出，这样可能会更好些。

armink, 2019-03-22:
好的，那等 v4.1 左右，我把 GC 开放出来

shuhuahuang, 2019-03-24:
期待能够在项目上用上该特性。

ZanoZ, 2019-04-09:
你好，我这几天正好在看这方面 ，刚刚从eeworld的老帖子翻到这里。首先感谢作者的开源。这个项目对于需要长时间高可靠性保存设定是非常有帮助的。
看了一下代码，对于扇区清理问题，因为这方面我之前也有一些模型上的思考，我认为在解决剩余扇区不多的问题上，可以有3个办法：
1，写操作完成并且剩余可用扇区不多的情况下(建议50%)，就触发清理一个脏扇区。
当然要确保后续写不会马上发生，如果万一发生了，要有对应的暂存机制，或者确保清除一个扇区的时间延迟可以接受。如果不能接受，就要冒险暂存在内存中，等待可能的x00mS时间延迟
2，开机或者关机时候集中清理，如果开机/关机会比较常发生，那么可以一次清理一部分，这个给应用选择。
3，给应用确定的清理函数，在实际应用中，在确保不会写情况下，比如采样还未开始，或者预计短时间内不会开始，或者刚刚开始采样预计确定时间内不会有数据要写，就可以调用清理，每次清理一部分，看看还未开始，再调用清理一下，分为若干次调用完成所有清理。

armink, 2019-04-10:
> 你好，我这几天正好在看这方面 ，刚刚从eeworld的老帖子翻到这里。首先感谢作者的开源。这个项目对于需要长时间高可靠性保存设定是非常有帮助的。
> 看了一下代码，对于扇区清理问题，因为这方面我之前也有一些模型上的思考，我认为在解决剩余扇区不多的问题上，可以有3个办法：
> 1，写操作完成并且剩余可用扇区不多的情况下(建议50%)，就触发清理一个脏扇区。 [...]

多谢你的建议，后面会先开放出来 GC API，开关机时 GC 你感觉会在哪些场景下需要

ZanoZ, 2019-04-21:
> 多谢你的建议，后面会先开放出来 GC API，开关机时 GC 你感觉会在哪些场景下需要

关机GC实际上可能是在mcu控制关机情况下，比如用户按关机按钮，然后mcu关闭显示关闭各项输出控制，然后最后(可能)关闭主mcu供电或者什么情况下，可以稍晚若干mS秒，先做一次检查看是否需要GC。
有些设备虽然用户操作关机，实际上是mcu仍旧在工作，只不过主输出或者显示屏都已经关闭，这时候是确定没啥事情会发生情况，那么也可以安安心心做GC的。
另外有些设备开机会有数秒时间，比如mcu操作外部供电刚上电，需要等待数秒稳定，或者显示欢迎界面什么的，这时候也是可以安安心心做GC的。

armink, 2019-07-07:
hi all, 最近对 GC 速度做了些提升，效果还是挺明显的，欢迎大家试用最新代码。

armink, 2019-08-23:
正式版已发布，可以再试试正式版。

closed 2019-08-23 by armink