armink/EasyFlash #44: 能支持C结构体数组吗？

chenguanfu511, 2019-03-20:
使用了API ef_set_struct和ef_get_struct，请问这个支持结构体数组吗

armink, 2019-03-20:
直接用 EasyFlash 4.0 版本，能够支持任何类型，无需再使用 types 插件

chenguanfu511, 2019-03-20:
看到示例代码，可以直接用UserPassword[5] userpassword存储这样吗
//typedef struct UserPassword {
//   uint8_t type;
//   uint8_t id;
//   uint8_t password;
//} UserPassword;

// 密码用户，从json->struct
void *user_password_get_cb(cJSON* json_obj) {
    s2j_create_struct_obj(struct_user_password, UserPassword);
    /* 反序列化数据到结构体对象 */
//    s2j_struct_get_basic_element(struct_user_password, json_obj, int, type);
    s2j_struct_get_basic_element(struct_user_password, json_obj, int, id);
    s2j_struct_get_basic_element(struct_user_password, json_obj, int, password);  // password长度为long
    /* 反序列化数据到结构体对象 */
    return struct_user_password;
}

cJSON *user_password_set_cb(void* struct_obj) {
    UserPassword *struct_user_password = (UserPassword *)struct_obj;
    /* 创建JSON对象 */
    s2j_create_json_obj(json_obj);
    /* 序列化数据到JSON对象 */
//    s2j_json_set_basic_element(json_obj, struct_user_password, int, type);
    s2j_json_set_basic_element(json_obj, struct_user_password, int, id);
    s2j_json_set_basic_element(json_obj, struct_user_password, int, password);
    return json_obj;
}

chenguanfu511, 2019-03-20:
解决了，感谢回复，工具很好用！！！！
```
size_t get_user_password_list_len() {
    size_t len=0;
    ef_get_env_blob("user_password_list", NULL, 0, &len);
    printf("len:%d\r\n", len);
    return len;
}

void get_user_password_list(UserPassword* user_password_list) {
    if(user_password_list!=NULL) {
//            user_keys_global = (char* ) pvPortMalloc(sizeof(char)*len);  // 务必释放资源
        ef_get_env_blob("user_password_list", user_password_list, sizeof(user_password_list), NULL);
        printf("user_password_list:%s\r\n", user_password_list);
    } else {
    }
}

void set_user_password_list(UserPassword* user_password_list) {
    if(user_password_list!=NULL) {
//            user_keys_global = (char* ) pvPortMalloc(sizeof(char)*len);  // 务必释放资源
        ef_set_env_blob("user_password_list", user_password_list, sizeof(user_password_list), NULL);
        printf("user_password_list:%s\r\n", user_password_list);
    } else {
    }
}
        len = get_user_password_list_len();
        if(len>0) {
            UserPassword* user_password_list = (UserPassword* ) pvPortMalloc(sizeof(UserPassword)*len);

            get_user_password_list(user_password_list);
            for(int i=0; i<len; ++i) {
                printf("user_password_list[%d]:(id,password):(%d, %d)\r\n", i, user_password_list[i].id, user_password_list[i].password);
            }
            if(user_password_list!=NULL) {
                vPortFree(user_password_list);
            }
        } else {
            UserPassword user_password_list[3];
            user_password_list[0].id = 0;
            user_password_list[0].password = 12;
            user_password_list[1].id = 1;
            user_password_list[1].password = 10;
            set_user_password_list(user_password_list);
        }
```

closed 2019-03-20 by chenguanfu511