armink/EasyFlash #45: 增加int数组消耗的env size大小问题

chenguanfu511, 2019-03-21:
我存储int数组，每一次增加2 int的数据，但是消耗的size提示增加176bytes，为什么这么大？
下面是日志
```
main---in---
[Flash](easyflash/ef_env.c:1421) ENV start address is 0x08019000, size is 20480 bytes.

[Flash]EasyFlash V4.0.0 is initialize success.

[Flash]You can get the latest version on https://github.com/armink/EasyFlash .

EF_NO_ERR
iap_need_copy_app=0
iap_copy_app_size=0
stop_in_bootloader=0
device_id=1
boot_times=0
user_password_format_list=blob @0x080197C8 22bytes

mode: next generation
size: 2016/10240 bytes.
main---in---
[Flash](easyflash/ef_env.c:1421) ENV start address is 0x08019000, size is 20480 bytes.

[Flash]EasyFlash V4.0.0 is initialize success.

[Flash]You can get the latest version on https://github.com/armink/EasyFlash .

EF_NO_ERR
iap_need_copy_app=0
iap_copy_app_size=0
stop_in_bootloader=0
device_id=1
boot_times=0
user_password_format_list=blob @0x08019878 24bytes

mode: next generation
size: 2192/10240 bytes.
```

armink, 2019-03-21:
你是用的什么平台？写入的粒度是多少？kv 的 name 有多长？

chenguanfu511, 2019-03-21:
平台是stm32f103c8t6，写入的粒度为(#define EF_WRITE_GRAN 32)，kv的name之前觉得是长度的原因，将修改为0基本没啥差别，(#define USER_PASSWORD_FORMAT_LIST_KEY "user_password_format_list")，实际中定义#define ENV_AREA_SIZE (2 * EF_ERASE_MIN_SIZE) 时，能写入216长度int类型的数组，那相当于2KB的flash利用了1KB，这怎么减小呢？

armink, 2019-03-21:
首先，写粒度是 32 bit 肯定会占用更多的空间，你可以认为每一种新的状态都要花费 4 个字节存储。参考设计文档（https://github.com/armink/EasyFlash/blob/master/docs/zh/design.md）可知，KV 的存储状态就得需要 20 字节。你可以执行下 resetenv ，单独算算加一个新的 env 到底开销多大。

其次， 216 长度数组 你 sizeof 一下就得将近 1k 了，不清楚你想在哪里减少？

chenguanfu511, 2019-03-21:
好的，就是mode: next generation size: 2192/10240 bytes.其中的size让我有点困惑～～

closed 2019-03-21 by chenguanfu511