armink/EasyFlash #48: ef_port_read 读出来的数据返回为-1，这个“-1”究竟代表什么呢？

zengwangfa, 2019-03-26:
ef_port_read(Nor_FLASH_ADDRESS+4*i,(u32 *)(&Normal_Parameter[i]),4);
[image]

armink, 2019-03-26:
贴出更多代码吧

zengwangfa, 2019-03-26:
**这个地址我还未写入数据，但是我想知道：**
还未写入数据的地址，如何去读取，返回的数据应该是什么样的数据？

**因为我读取Flash设置了一个简单的自检去判断 数据是否正确，当判断数据 不为（-1）时则为正常数据，但是可能这个 打印出来的（-1）并不是真正的 数值上的（-1），这个简单的判断并不能成立。**
```c
/* 读取 FLASH 普通参数*/
int Normal_Paramter_Init_With_Flash(void)
{
		char i = 0;
		for(i = 0;i < PARAMEMER_MAX_NUMBER_A;i++ ){
				ef_port_read(Nor_FLASH_ADDRESS+4*i,(u32 *)(&Normal_Parameter[i]),4);		
		}
		
		Normal_Parameter_SelfCheck_With_Flash();//Flash参数自检 若为 0 或 -1则为 非正常数据，则不传递给真实数据 

		log_i("Flash_Read()");
		return 0;
}
```

armink, 2019-03-26:
- 1、 ef_port_read 不是给用户调用的，是由用户适配好，提供给 EasyFlash 使用的
- 2、即便你去调用这个接口，他的出参是 EfErrCode 类型，不应该是 -1

https://github.com/armink/EasyFlash/blob/f64af09c2271dc8192f06d8856f54f87b034bf3b/easyflash/port/ef_port.c#L64-L71

closed 2019-08-23 by armink