armink/EasyFlash #51: [SFUD]Error: Flash address is out of bound.

qinhao4832, 2019-04-15:
[SFUD]Error: Check SFDP signature error. It's must be 50444653h('S' 'F' 'D' 'P').
[SFUD]Warning: Read SFDP parameter header information failed. The SST25VF016B is not support JEDEC SFDP.
[SFUD]Warning: This flash device is not found or not support.
[SFUD]Error: SST25VF016B flash device is initialize fail.
[Flash](..\Src\easyflash\src\ef_env.c:1419) ENV start address is 0x00000000, size is 819200 bytes.
[SFUD]Error: Flash address is out of bound.
[Flash]Warning: Sector header check failed. Set it to default.
[SFUD]Error: Flash address is out of bound.
[SFUD]Error: Flash address is out of bound.
[Flash]EasyFlash V4.0.0 is initialize success.

测试环境
STM32F207 外挂flash。
2块板，A板跑同样程序，没有异常。 B板，提示上述错误。
可以排除B板的SPI flash硬件有问题。因为另外一套程序（没有使用EasyFlash），在B板上运行，可以正常读写SPI flash。

qinhao4832, 2019-04-15:
哦，知道了。读不到 JEDEC ID，  flash->chip.capacity默认就是0、

qinhao4832, 2019-04-15:
[SFUD](..\Src\sfud\src\sfud.c:104) Start initialize Serial Flash Universal Driver(SFUD) V0.10.29.
[SFUD](..\Src\sfud\src\sfud.c:105) You can get the latest version on https://github.com/armink/SFUD .
[SFUD](..\Src\sfud\src\sfud.c:748) The flash device manufacturer ID is 0xEF, memory type ID is 0x40, capacity ID is 0x15.
[SFUD]Error: Check SFDP signature error. It's must be 50444653h('S' 'F' 'D' 'P').
[SFUD]Warning: Read SFDP parameter header information failed. The SST25VF016B is not support JEDEC SFDP.
[SFUD]Warning: This flash device is not found or not support.
[SFUD]Error: SST25VF016B flash device is initialize fail.
[Flash](..\Src\easyflash\src\ef_env.c:1419) ENV start address is 0x00000000, size is 819200 bytes.
[SFUD](..\Src\sfud\src\sfud.c:306) (flash->init_ok) has assert failed at sfud_read.

armink, 2019-04-15:
SFUD 都没有初始化成功呀，确定下 Flash 型号

closed 2019-08-23 by armink