armink/EasyFlash #55: default_env_set只能保存字符串？

tianxiaoMCU, 2019-04-25:
4.0支持任意类型的值，default_env_set如何保存非字符串类型？

armink, 2019-04-25:
可以是其他类型，default_env_set 结构体第二个入参类型是 void*，第三个参数是长度

tianxiaoMCU, 2019-04-28:
默认环境变量设置如下
```
static const ef_env default_env_set[] = {
    {"para1", "123", 4},
    {"para2", "456", 4},
    {"para3", "789", 4},
};
```
调用`easyflash_init()`初始化成功，但是调用`ef_get_env_blob("para1", buf, sizeof(buf), NULL);`获取变量失败，返回结果0

armink, 2019-04-28:
Flash 中的默认环境变量只能存储一次，如果之前程序运行时候存储过了，下次程序再运行是不会执行恢复默认的操作。

你手动调用恢复默认环境变量 API 后再试试

tianxiaoMCU, 2019-04-29:
```
  if (easyflash_init() != EF_NO_ERR)
  {
    log_e("Init easyflash failed");
  }
  char buf[5] = {0};
  size_t len = 0;
  len = ef_get_env_blob("para1", buf, sizeof(buf), NULL);
  if (0 == len)
  {
    log_e("para not found");
  }
  else
  {
    log_d("para = %s", buf);
  }
```
我是这么用的，初始化后就去读出来。下面是读出的flash的数据，只在两个Page（STM32L431CC，只用了最后的两个page）的开始处有数据，但是没发现有默认数据，应该是没存储过。
[image]
[image]

armink, 2019-04-29:
L  系列的注意下写粒度，记得是 64 bit。

你可以单步看下 ef_env_set_default 的执行过程

tianxiaoMCU, 2019-04-29:
```
/* the flash write granularity, unit: bit
 * only support 1(nor flash)/ 8(stm32f4)/ 32(stm32f1)/ 64(stm32l4) */
#define EF_WRITE_GRAN             64/* @note you must define it for a value */
```
注释里有提示，也查了手册，所以保证是64bit。

单步 ef_env_set_default 的执行过程，'create_env_blob'函数中`update_sec_status`时写（地址0x0803F008）失败。但是`ef_env_set_default`中没有获取`create_env_blob`的返回值，而是直接返回`EF_NO_ERR`，后面也就提示初始化成功了。

`ef_port_read`、`ef_port_erase`、`ef_port_write`单独测试暂时没发现问题。

armink, 2019-04-29:
试试手动操作下 update_sec_status 操作失败的地址及数据内容呢

tianxiaoMCU, 2019-04-29:
先把page擦除，调用`write_status(0x0803F008, status_table, SECTOR_STORE_STATUS_NUM, SECTOR_STORE_USING);`（update_sec_status 里也是调的这个），结果OK

armink, 2019-04-29:
page 擦除在第一次 format 时执行过了，这样后续的状态都能顺序的写入到 flash page 头部了

closed 2019-08-23 by armink