armink/EasyFlash #57: 关于写入速度

TeemoRZ, 2019-04-29:
无论写入多大的数据，延时都大于2s，尝试修改retry_delay_100us  有所减少，但还是很慢？是否有对应的数据呢》
MCU:STM32F105RCT6
FREQUENCE:72MHZ
FLASH CHIP:W25Q128

移植的读写接口如下：
static sfud_err spi_write_read(const sfud_spi *spi, const uint8_t *write_buf, size_t write_size, uint8_t *read_buf,
        size_t read_size) {
    sfud_err result = SFUD_SUCCESS;
    uint8_t send_data, read_data;
					int i = 0;
    spi_user_data_t spi_dev = (spi_user_data_t) spi->user_data;

    if (write_size) {
        SFUD_ASSERT(write_buf);
    }
    if (read_size) {
        SFUD_ASSERT(read_buf);
    }

    
		FLASH_SPI_CS_ENABLE();
    if (write_size && read_size)
    {
        /* read data */
			
			uint8_t dummy_byte = 0xff;
      //  qspi_send_then_recv(write_buf, write_size, read_buf, read_size);
			for( i = 0; i < write_size; i++)
			{
				//HAL_SPI_TransmitReceive(&hspi2, (uint8_t*)write_buf+i, &dummy_byte, 1,0xFFFFFF);
				SPI_FLASH_SendByte(write_buf[i]);
			}
			for(i = 0; i < read_size; i++)
			{
					//HAL_SPI_TransmitReceive(&hspi2, &dummy_byte, read_buf, 1,0xFFFFFF);
				read_buf[i]=SPI_FLASH_SendByte(dummy_byte);
			}
    }
    else if (write_size)
    {
        /* send data */
       // qspi_send_then_recv(write_buf, write_size, NULL, NULL);
			HAL_SPI_Transmit(&hspi2,(uint8_t *) write_buf, write_size,0xFFFFFF);

    }

		FLASH_SPI_CS_DISABLE();
    return result;
}

TeemoRZ, 2019-04-29:
补充一下LOG ，可以看到 写一个时长要3s多

[16:43:38.726]收←◆[Flash](..\KV\easyflash\ef_env.c:1001) Trigger a GC check after created ENV.

[16:43:38.936]收←◆[Flash](..\KV\easyflash\ef_env.c:907) The remain empty sector is 948, GC threshold is 1.
214
[16:43:42.333]收←◆215
[16:43:45.726]收←◆216
[16:43:49.120]收←◆217
[16:43:52.517]收←◆218
[16:43:55.918]收←◆219
[16:43:59.320]收←◆220
[16:44:02.723]收←◆221
[16:44:06.131]收←◆222
[16:44:09.542]收←◆223
[16:44:12.956]收←◆[Flash](..\KV\easyflash\ef_env.c:1001) Trigger a GC check after created ENV.

[16:44:13.166]收←◆[Flash](..\KV\easyflash\ef_env.c:907) The remain empty sector is 948, GC threshold is 1.
224
[16:44:16.576]收←◆225
[16:44:19.980]收←◆226
[16:44:23.388]收←◆227
[16:44:26.799]收←◆228
[16:44:30.211]收←◆229
[16:44:33.627]收←◆230
[16:44:37.044]收←◆231
[16:44:40.464]收←◆232
[16:44:43.888]收←◆233
[16:44:47.315]收←◆[Flash](..\KV\easyflash\ef_env.c:1001) Trigger a GC check after created ENV.

[16:44:47.525]收←◆[Flash](..\KV\easyflash\ef_env.c:907) The remain empty sector is 948, GC threshold is 1.
234
[16:44:50.949]收←◆235

armink, 2019-04-29:
会不会跟 SPI 驱动有关系，先转一下 SPI 波形分析一下吧

TeemoRZ, 2019-04-29:
> 会不会跟 SPI 驱动有关系，先抓一下 SPI 波形分析一下吧

这个SPI驱动和之前 直接驱动flash是同一个驱动，是直接的hal库 api；
之前写入速度几乎在100ms左右

TeemoRZ, 2019-04-29:
> > 会不会跟 SPI 驱动有关系，先抓一下 SPI 波形分析一下吧
> 
> 这个SPI驱动和之前 直接驱动flash是同一个驱动，是直接的hal库 api；
> 之前写入速度几乎在100ms左右

另外一个发现，随着写入次数的增加，写入的速度越来越慢

TeemoRZ, 2019-04-29:
是否有相关的测试benchmark？

armink, 2019-04-29:
4.0 在减少 RAM 占用的条件下牺牲了环境变量操作的速度，如果对速度要求苛刻，可以先使用 3.0 版本。

接下来会对 4.0 的速度问题重点优化，如果兴趣欢迎加入进来一起优化。

TeemoRZ, 2019-04-29:
我调试了一下发现，每次都会执行如下代码，检测 所有env，这里消耗大部分时间
        /* search all ENV */
            while ((env->addr.start = get_next_env_addr(&sector, env)) != FAILED_ADDR) {
                read_env(env);
                /* iterator is interrupted when callback return true */
                if (callback(env, arg1, arg2)) {
                    return;
                }
            }

ENV 缓存在ram中还是有必要

TeemoRZ, 2019-04-29:
建议用类型文件系统中的INODE ,维护一张表用来映射数据的地址；alios things  有类型的 K - V storage 机制，不知是不是参考你的项目，本想移植alios 但依赖其本身系统太多，不太好移植

armink, 2019-04-29:
嗯，这个是一个优化点，你可以先改下试试。

armink, 2019-07-02:
hi, @TeemoRZ ，最新的版本新加入了缓存机制，查找、写入效率提升非常明显，你可以试试哈~~

closed 2019-08-23 by armink