armink/EasyFlash #62: When will be V4.0 released ?

ShareCat, 2019-05-22:
When will be V4.0 released ?
We are looking forward to it, TKS !

armink, 2019-05-22:
I am optimizing the query efficiency of env now. Other features are stable. It will be released as early as the end of June.

ZanoZ, 2019-06-21:
所以4.0现在还没出来是吧.

armink, 2019-06-21:
> 所以4.0现在还没出来是吧.

稍安勿躁，正式版还未发布

ZanoZ, 2019-06-21:
哇,这么近,上班了?加个微信吧28277961

ZanoZ, 2019-06-26:
https://github.com/ZanoZ/EEFile
我忍不住开始动手写了,不过我觉得还是有点浪费,大家造同一个轮子重复劳动啊....( ･᷄ὢ･᷅ )

armink, 2019-06-26:
> https://github.com/ZanoZ/EEFile
> 我忍不住开始动手写了,不过我觉得还是有点浪费,大家造同一个轮子重复劳动啊....( ･᷄ὢ･᷅ )

你的链接打不开~~

跟大家同步下进展，上周末针对 env 及 sector 完成了缓存机制的开发，保证了安全性的前提下提升了效率，测试结果还不错，估计这周末能同步上来。

效率的问题解决后，其他剩余的小问题就很少了。

hzm476172557, 2019-06-27:
you have done a great job, I think the easyflash 5 should do some subtraction. It will be better if you abandon the old APIs

armink, 2019-07-02:
hi, all. 最近新版本新加入了缓存机制，查询，写入效率得到大幅提升，欢迎大家测试~~

目前，V4.0 基本没有待完善的功能了。等大家对缓存功能有更多的测试反馈后，近期就将正式发布。

ZanoZ, 2019-07-03:
> > https://github.com/ZanoZ/EEFile
> > 我忍不住开始动手写了,不过我觉得还是有点浪费,大家造同一个轮子重复劳动啊....( ･᷄ὢ･᷅ )
> 
> 你的链接打不开~~
> 
> 跟大家同步下进展，上周末针对 env 及 sector 完成了缓存机制的开发，保证了安全性的前提下提升了效率，测试结果还不错，估计这周末能同步上来。
> 
> 效率的问题解决后，其他剩余的小问题就很少了。

写了一半,感觉太不成熟就先设成不公开了,我加你进去,还请指点哈.

armink, 2019-08-23:
正式版已经发布了，欢迎使用

closed 2019-08-23 by armink