armink/EasyFlash #64: GC出现问题，未掉电，中等强度持续跑12小时

Flymachine, 2019-06-13:
env区：24K；log区：24K；ARM平台，FreeRTOS，移植，使用master最新版本的EasyFlash。

在测试时`wifi_ssid`、`wifi_pwd`、`__ver_num__`均保持初始值。只有`l_unup_i`保持定期写入新值。

距离错误最近一次GC搬移，注意`0x080F0039`（以下日志均发生在写入`l_unup_i`新值时）：
```
[Flash](ef_env.c:1000) Trigger a GC check after created ENV.
[Flash](ef_env.c:906) The remain empty sector is 1, GC threshold is 1.
[Flash](ef_env.c:821) Moved the ENV (wifi_ssid) from 0x080F0014 to 0x080F5014.
[Flash](ef_env.c:821) Moved the ENV (wifi_pwd) from 0x080F0039 to 0x080F5039.
[Flash](ef_env.c:821) Moved the ENV (__ver_num__) from 0x080F0061 to 0x080F5061.
[Flash](ef_env.c:886) Collect a sector @0x080F0000
[Flash](ef_env.c:886) Collect a sector @0x080F1000
[Flash](ef_env.c:886) Collect a sector @0x080F2000
[Flash](ef_env.c:886) Collect a sector @0x080F3000
[Flash](ef_env.c:821) Moved the ENV (l_unup_i) from 0x080F4FC0 to 0x080F5088.
[Flash](ef_env.c:886) Collect a sector @0x080F4000
```
出错时：
1. 
```
[Flash](ef_env.c:336) Error: The ENV @0x080F004E length has an error.
```
2. 
```
[Flash](ef_env.c:726) Trigger a GC check after alloc ENV failed.
[Flash](ef_env.c:837) Warning: Alloc an ENV (size 34) failed when new ENV. Now will GC then retry.
[Flash](ef_env.c:906) The remain empty sector is 1, GC threshold is 1.
[Flash](ef_env.c:886) Collect a sector @0x080F0000
[Flash](ef_env.c:821) Moved the ENV (wifi_ssid) from 0x080F5014 to 0x080F0014.
[Flash]Error: The ENV (@0x080F0039) CRC32 check failed!
[Flash](ef_env.c:336) Error: The ENV @0x080F0088 length has an error.
[Flash](ef_env.c:821) Moved the ENV (wifi_pwd) from 0x080F5039 to 0x080F4014.
[Flash]Error: The ENV (@0x080F0039) CRC32 check failed!
[Flash](ef_env.c:336) Error: The ENV @0x080F406C length has an error.
[Flash](ef_env.c:336) Error: The ENV @0x080F4084 length has an error.
[Flash](ef_env.c:881) Error: Moved the ENV (__ver_num__) for GC failed.
[Flash](ef_env.c:886) Collect a sector @0x080F5000
[Flash]Error: The ENV (@0x080F0039) CRC32 check failed!
[Flash](ef_env.c:726) Trigger a GC check after alloc ENV failed.
```
3.
```
[Flash]Error: The ENV (@0x080F0039) CRC32 check failed!
[Flash](ef_env.c:336) Error: The ENV @0x080F504E length has an error.
```
之后每次写入env都会报：
```
[Flash]Error: The ENV (@0x080F0039) CRC32 check failed!
```

`0x080F0039`是`wifi_pwd`在GC搬移前的老位置，GC后应当失效。且程序从未重新设置`wifi_pwd`过，运行中也从未掉电，供电也是独立供电，sem信号锁已设置，系统内部还有专门的flash锁。这种问题会是什么问题？

ZanoZ, 2019-06-14:
周边信息太少很难预估问题方向啊

Flymachine, 2019-06-14:
回头我用V3.3试试。
芯片是W600，龙哥说是那段日志是alloc env空间出错引发GC，可能和驱动有关系，但他没能复现。
发自我的iPhone

armink, 2019-06-15:
W600 建议先用 V3.3 版本，底层 flash 驱动经测试发现，写入时中途掉电，还存在扇区数据全部丢失的问题，可能与 xip 的 cache 有关。所以 V4.0 可能暂时在 W600 片上 flash 上存在问题，后续我再跟原厂深入确认下吧。

Flymachine, 2019-06-20:
> 
> 
> W600 建议先用 V3.3 版本，底层 flash 驱动经测试发现，写入时中途掉电，还存在扇区数据全部丢失的问题，可能与 xip 的 cache 有关。所以 V4.0 可能暂时在 W600 片上 flash 上存在问题，后续我再跟原厂深入确认下吧。

是的，昨天测试W600时EasyFlash V4.0又出现了`EF_ASSERT(0)`错误，经查是跳入了`//TODO 扇区连续模式，或者写入长度没有写入完整`块里了，应该就是数据没写完。
V3.3经过12小时测试倒是没有问题。此外，另一块M4架构的芯片用V4.0进行同样的测试也没有问题。
W600的flash驱动确实有点问题。
到此结贴了，W600官方SDKv3.2.0下就用EasyFlash V3.3就行了。

closed 2019-06-20 by Flymachine