armink/EasyFlash #65: The WL mode in EFV3.3.0 is not working?

hzm476172557, 2019-06-28:
Hey,dude:
I still using the version of 3.3.0 that you have already released.
I am trying to use WL mode,and I have defined EF_ENV_USING_WL_MODE in ef_cfg.h.But it seems does't work.
I checked the memory address,it still write the same address every time when I call ef_save_env().
Am I missing something?

hzm476172557, 2019-06-28:
/* the user setting size of ENV, must be word alignment */
#define ENV_USER_SETTING_SIZE     320
/* ENV area total bytes size in normal mode. */
#define ENV_AREA_SIZE             (3 * EF_ERASE_MIN_SIZE)  

besides,
Up 2 defines,I don't get their means well.I set  3 * EF_ERASE_MIN_SIZE for ENV_AREA_SIZE,but it always using the section 2 of the area.I still don't get it (T_T)

hzm476172557, 2019-07-01:
One more question:
If the device neet to reset to factory,How to reset the env?
Just need to call ef_env_set_default()?

armink, 2019-07-01:
yes, you need to see the documents first

[image]

closed 2019-07-02 by armink