armink/EasyFlash #70: V4.0.0版本移植到Apollo2出现hardfault

tianxiaoMCU, 2019-08-23:
MCU：apollo2
SDK版本：V1.2.12
环境：keil V5.26.2

移植V4.0.0后出现hardfault，使用cmbacktrace定位，数据如下
[image]
addr2line定位结果：
[image]
根据定位结果找到
[image]
**但是这个函数没有被调用**

**单步调试，执行下图红框处后进入hardfault**
[image]

armink, 2019-08-23:
cmb 显示的问题估计跟移植有关系，要确保 C 主栈等名称配置正确。

handfault 可以尝试看下单步的汇编指令情况，单独看你的截图没有看出有啥问题。

tianxiaoMCU, 2019-08-24:
解决了，确实跟移植有关系。记错了EF_WRITE_GRAN的单位，拿来算word的数量导致的

closed 2019-08-24 by tianxiaoMCU