armink/EasyFlash #71: 4.0.0初始化提示成功，但是保存参数失败

tianxiaoMCU, 2019-08-24:
**单独测试移植的擦除、读写接口是成功的**
初始化提示成功，但是读取默认参数失败，找不到这个参数，保存参数也失败
[image]
默认参数如下
```
static uint32_t boot_count = 0;
static uint32_t boot_time[10] = {0, 1, 2, 3};

/* default environment variables set for user */
static const ef_env default_env_set[] = {
    {"username", "armink", 0},
    {"boot_count", &boot_count, sizeof(boot_count)},
    {"boot_time", &boot_time, sizeof(boot_time)},
};
```
测试代码如下
```
static void test_env(void) {
  uint32_t i_boot_times = NULL;
  char *c_old_boot_times, c_new_boot_times[11] = {0};
  int32_t i32ReturnCode;

  /* get the boot count number from Env */
  c_old_boot_times = ef_get_env("boot_time");

  i_boot_times = atol(c_old_boot_times);
  /* boot count +1 */
  i_boot_times++;
  am_util_stdio_printf("The system now boot %d times\n\r", i_boot_times);
  /* interger to string */

  sprintf(c_new_boot_times, "%ld", i_boot_times);
  /* set and store the boot count number to Env */
  i32ReturnCode = ef_set_env("boot_time", c_new_boot_times);
  if (i32ReturnCode != EF_NO_ERR)
  {
    am_util_stdio_printf("Set env \"boot_time\" failed, error code = %d\r\n", i32ReturnCode);
  }

  i32ReturnCode = ef_save_env();
  if (i32ReturnCode != EF_NO_ERR)
  {
    am_util_stdio_printf("Save env failed, error code = %d\r\n", i32ReturnCode);
  }
}
```

armink, 2019-08-26:
什么平台及 flash 呢。应该是移植没有对接好的，请检查配置文件及移植代码，也可以优先在 demo 平台上测试下相关代码

tianxiaoMCU, 2019-08-26:
apollo2的内部flash

zrw269113179, 2019-10-09:
这问题我也出现了，后来查出是ef_load_env()中的sector.combined没有初始化导致的，加一句sector.combined = 0xffffffff;就好了。这问题只会在第一次写入扇区出现，后面都不会。
测试环境stm32f103+w25q16。然后还会出现环境变量初始化导致rtthread进入死循环，把ef_env_set_default()放到ef_env_init()最下方等初始化完成再调用就好了。

armink, 2019-10-09:
> 这问题我也出现了，后来查出是ef_load_env()中的sector.combined没有初始化导致的，加一句sector.combined = 0xffffffff;就好了。这问题只会在第一次写入扇区出现，后面都不会。
> 测试环境stm32f103+w25q16。然后还会出现环境变量初始化导致rtthread进入死循环，把ef_env_set_default()放到ef_env_init()最下方等初始化完成再调用就好了。

试试最新版呢，sector 数值问题已经修正了，详见： https://github.com/armink/EasyFlash/pull/73/files#diff-a8e474d4f098988ab755c2e37b01da1aR622

死循环的问题有点不明白，可否详细说说