armink/EasyFlash #75: 关于存储大量参数出现的错误

cmhgithub, 2019-10-09:
static void test_env(void) {
	char test_key[64];
	static int test_value = 0;
	
	static int test_read_value = 0;
	int value;
	
	if(test_value < 2000){
		sprintf(test_key, "target%d", test_value);
		if(EF_NO_ERR != ef_set_int(test_key, test_value)){
			printf("wite %s = %d;ERROR!!!!!!!!!!\n\r", test_key, test_value);
			delay(6000000);
			delay(6000000);
			delay(6000000);
		}else{
			printf("wite %s = %d;\n\r", test_key, test_value);
		}
		test_value++;
	}else{
		sprintf(test_key, "target%d", test_read_value);
		value = ef_get_int(test_key);
		printf("read %s = %d;\n\r", test_key, value);
		test_read_value++;
	}
}
使用以上代码大量存储数据时发现一个问题，存储的数据读取出来后出现错误，读取时对应不上。且存储过程中并没有报错。少量的存储测试都是正常的。

armink, 2019-10-10:
改进了一下你的测试用例如下

```c
static void ef_bench_test(uint8_t argc, char **argv) {
    if (argc >= 2) {
        char test_key[6 + 10];
        size_t i = 0, saved_i, count = atoi(argv[1]);
        while (i < count) {
            snprintf(test_key, sizeof(test_key), "target%d", i);
            if (EF_NO_ERR != ef_set_env_blob(test_key, &i, sizeof(i))) {
                rt_kprintf("wite %s = %d;ERROR!!!!!!!!!!\r\n", test_key, i);
                break;
            }
            i ++;
        }
        i = 0;
        /* read check */
        while (i < count) {
            snprintf(test_key, sizeof(test_key), "target%d", i);
            if (0 == ef_get_env_blob(test_key, &saved_i, sizeof(saved_i), NULL) || i != saved_i ) {
                rt_kprintf("read %s != %d;ERROR!!!!!!!!!!\r\n", test_key, saved_i);
                break;
            }
            i++;
        }
    }
}
MSH_CMD_EXPORT(ef_bench_test, EasyFlash bechmark test);
```

测试了 2000 次没有发现问题，建议：
- 检查驱动软件及硬件状态
- 更换为 blob 接口进行测试

测试日志如下：

```shell
msh />ef_bench_test 2000
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 39, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 38, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 37, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 36, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 35, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 34, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 33, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 32, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 31, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 30, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 29, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 28, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 27, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 26, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 25, GC threshold is 1.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1326) Trigger a GC check after created ENV.
[Flash] (../packages/EasyFlash-latest/src/ef_env.c:1222) The remain empty sector is 24, GC threshold is 1.
[... 2002 lines cut ...]
target1980=blob @0x000128A8 4bytes
target1981=blob @0x000128CE 4bytes
target1982=blob @0x000128F4 4bytes
target1983=blob @0x0001291A 4bytes
target1984=blob @0x00012940 4bytes
target1985=blob @0x00012966 4bytes
target1986=blob @0x0001298C 4bytes
target1987=blob @0x000129B2 4bytes
target1988=blob @0x000129D8 4bytes
target1989=blob @0x000129FE 4bytes
target1990=blob @0x00012A24 4bytes
target1991=blob @0x00012A4A 4bytes
target1992=blob @0x00012A70 4bytes
target1993=blob @0x00012A96 4bytes
target1994=blob @0x00012ABC 4bytes
target1995=blob @0x00012AE2 4bytes
target1996=blob @0x00012B08 4bytes
target1997=blob @0x00012B2E 4bytes
target1998=blob @0x00012B54 4bytes
target1999=blob @0x00012B7A 4bytes

mode: next generation
size: 76127/159744 bytes.
msh />
```

armink, 2019-10-12:
如果没有问题，这个 issue 先关闭了

closed 2019-10-12 by armink

cmhgithub, 2020-02-21:
关于这个问题最近再次进行了测试，使用的附带的demo。单片机stm32f103VET6，路径EasyFlash-master\demo\env\stm32f10x\rtt\。修改ENV_AREA_SIZE (150 * EF_ERASE_MIN_SIZE) 也就是300k，（EF_ERASE_MIN_SIZE 为2048）。在全新系统环境下，第一次测试不会出现问题。第二次输入命令，必然会在target109处出错。另外我自己使用ram模拟flash的上位机也会有同样的现象，驱动、硬件问题的可能性应该不大。
测试日志 
: command not found.
msh >
msh >ef
ef_bench_test
msh >ef_bench_test 500
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             49, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             48, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             47, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             46, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             45, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             44, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             43, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             42, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             41, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after cre                                                                                                                                                             ated ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 1                                                                                                                                                             40, GC threshold is 1.
[... cut ...]
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 124, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after created ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 123, GC threshold is 1.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1362) Trigger a GC check after created ENV.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1258) The remain empty sector is 122, GC threshold is 1.
read target109 != 10;ERROR!!!!!!!!!!
msh >