armink/EasyFlash #77: 内存对齐访问的问题

mojinpan, 2019-10-27:
在函数:static EfErrCode create_env_blob(sector_meta_data_t sector, const char *key, const void *value, size_t len)
中的result = align_write(env_addr + ENV_HDR_DATA_SIZE, (uint32_t *) key, env_hdr.name_len);
此处将key 是将8位地址强制转换成32位地址,后面读取时可能存在内存字节对齐的问题

armink, 2019-10-27:
嗯，是有可能的，你用是什么平台

mojinpan, 2019-10-27:
芯片是:STM32F030CC
下级函数暂时先做特殊处理:
uint32_t StmFlashWriteby16bit(uint32_t addr,uint16_t *buf,size_t size)
{	
	uint32_t addrx=0;
	uint32_t endaddr=0;	
	uint16_t data =0;
	uint8_t *pdata;
	pdata = (uint8_t*)buf;
			
    if(addr<STM32_FLASH_START_ADDR || addr + size > STM32_FLASH_END_ADDR || addr % 2)
	{
		return 0 ;//非法地址
	}
	
 	HAL_FLASH_Unlock();//解锁	  
	for(addrx=addr,endaddr=addr+size; addrx<endaddr; addrx+=2,pdata +=2)//写数据
	{
		data = *pdata | (*(pdata+1))<<8;//防止内存对齐出错
//		if(HAL_FLASH_Program(FLASH_TYPEPROGRAM_HALFWORD,addrx,*buf)!=HAL_OK)//这样有内存对齐的风险!
		if(HAL_FLASH_Program(FLASH_TYPEPROGRAM_HALFWORD,addrx,data)!=HAL_OK)//写入数据
		{ 
			addrx = addrx -2;
			ULOG_E("StmFlash wirte error code:0x%x\r\n",HAL_FLASH_GetError());
			break;	//写入异常				
		}
	} 
	HAL_FLASH_Lock();//上锁	
	
	return addrx - addr;
}

mojinpan, 2019-10-27:
应该所有的stm32的芯片都存在这个问题吧,好像内部flash没有能按8位写入的

mojinpan, 2019-10-27:
另外,我用cm_backtrace追踪的时候也没能保持内存对齐的错误,不知道是什么原因,追踪的时候返回:

```
shell>>env_test 235346 2363476

Firmware name: BC854, hardware version: V1.0.0, software version: V0.1.0
Fault on interrupt or bare metal(no OS) environment
===== Thread stack information =====
  addr: 20004ec8    data: 2000016d
  addr: 20004ecc    data: 00000004
  addr: 20004ed0    data: 0803c504
  addr: 20004ed4    data: 00000004
  addr: 20004ed8    data: 0803c500
  addr: 20004edc    data: 2000016d
  addr: 20004ee0    data: 00000000
  addr: 20004ee4    data: 0800baf9
  addr: 20004ee8    data: 00000000
  addr: 20004eec    data: 00000006
  addr: 20004ef0    data: 0803c500
  addr: 20004ef4    data: 2000016d
  addr: 20004ef8    data: 00000000
  addr: 20004efc    data: 0800a6d3
  addr: 20004f00    data: 00000004
  addr: 20004f04    data: ffffffff
  addr: 20004f08    data: 00000001
  addr: 20004f0c    data: 00000000
  addr: 20004f10    data: 0803c4d8
  addr: 20004f14    data: 20000a70
  addr: 20004f18    data: ffffffff
  addr: 20004f1c    data: 0800af4f
  addr: 20004f20    data: ffffffff
  addr: 20004f24    data: 000000ff
  addr: 20004f28    data: 00000000
  addr: 20004f2c    data: ffffff00
  addr: 20004f30    data: ffffffff
  addr: 20004f34    data: ffffffff
  addr: 20004f38    data: ffffffff
  addr: 20004f3c    data: ffffffff
  addr: 20004f40    data: 3034564b
  addr: 20004f44    data: 00000038
  addr: 20004f48    data: 76a5f634
  addr: 20004f4c    data: ffffff06
  addr: 20004f50    data: 00000007
  addr: 20004f54    data: 20000a70
  addr: 20004f58    data: 2000016d
  addr: 20004f5c    data: 20000174
  addr: 20004f60    data: 00000007
  addr: 20004f64    data: 20000174
  addr: 20004f68    data: 00000000
  addr: 20004f6c    data: 2000016d
  addr: 20004f70    data: 00000007
  addr: 20004f74    data: 0800cc7f
  addr: 20004f78    data: 00000006
  addr: 20004f7c    data: 00000000
  addr: 20004f80    data: 01010101
  addr: 20004f84    data: 2000016d
  addr: 20004f88    data: 20000174
  addr: 20004f8c    data: 00000007
  addr: 20004f90    data: 00000000
  addr: 20004f94    data: 0800bbbf
  addr: 20004f98    data: 01010101
  addr: 20004f9c    data: 20000174
  addr: 20004fa0    data: 2000016d
  addr: 20004fa4    data: 00000007
  addr: 20004fa8    data: 00000005
  addr: 20004fac    data: 0800bb91
  addr: 20004fb0    data: 20000198
  addr: 20004fb4    data: 00000003
  addr: 20004fb8    data: 080174a0
  addr: 20004fbc    data: 080022d1
  addr: 20004fc0    data: 0000000a
  addr: 20004fc4    data: 20000164
  addr: 20004fc8    data: 00000001
  addr: 20004fcc    data: 08000927
  addr: 20004fd0    data: 0000000a
  addr: 20004fd4    data: 08005031
  addr: 20004fd8    data: 0800d02a
  addr: 20004fdc    data: 0800ce37
  addr: 20004fe0    data: 20000164
  addr: 20004fe4    data: 20000164
  addr: 20004fe8    data: 00000003
  addr: 20004fec    data: 0800cff3
  addr: 20004ff0    data: 00000001
  addr: 20004ff4    data: 00000001
  addr: 20004ff8    data: 00000000
  addr: 20004ffc    data: 0000000d
  addr: 20005000    data: 20000164
  addr: 20005004    data: 00000000
  addr: 20005008    data: 00000000
  addr: 2000500c    data: 0800d27b
  addr: 20005010    data: 20000164
  addr: 20005014    data: 20000164
  addr: 20005018    data: 00000000
  addr: 2000501c    data: 0800d901
  addr: 20005020    data: 0000000d
  addr: 20005024    data: 08017374
  addr: 20005028    data: 20001fe8
  addr: 2000502c    data: 0800534d
  addr: 20005030    data: 0800f908
  addr: 20005034    data: 00000005
  addr: 20005038    data: 08017374
  addr: 2000503c    data: 08002095
  addr: 20005040    data: 08017374
  addr: 20005044    data: 0800c4e1
====================================
=================== Registers information ====================
  R0 : 0803c504  R1 : 40022000  R2 : 00000004  R3 : 00000000
  R12: 00000000  LR : 080051a7  PC : 080051ae  PSR: 81000000
==============================================================
Show more call stack info by run: addr2line -e BC854.axf -a -f 080051ae 080051a3 0800baf5 0800a6cf 0800af4b 0800cc7b 0800bbbb 0800bb8d 080022cd 08000923 0800502d 0800ce33 0800cfef 0800d277 0800d8fd 08005349 
```

armink, 2019-10-27:
其他一些芯片是支持的，F0 特殊一些

- 内存对齐问题，这个建议用户在 API 使用的时候，保证传入的 key name 符合自己平台的对齐要求
- Flash 对齐问题，这个目前不存在，这个 align_write 函数的目的是将数据，按对齐的长度写入对齐的地址

不知道这样能否解决你的问题

closed 2019-10-27 by mojinpan