armink/EasyFlash #80: 测试demo失败

yeweizhong, 2019-11-18:
hi,armink.
我在使用demo测试，把sfud移植完成后验证easyflash报错。

报错信息如下：
SFUD](..\components\sfud\src\sfud.c:104) Start initialize Serial Flash Universal Driver(SFUD) V0.10.29.
[SFUD](..\components\sfud\src\sfud.c:105) You can get the latest version on https://github.com/armink/SFUD .
[SFUD](..\components\sfud\src\sfud.c:748) The flash device manufacturer ID is 0xC8, memory type ID is 0x40, capacity ID is 0x16.
[SFUD]Find a GigaDevice GD25Q32CSIG flash chip. Size is 4194304 bytes.
[SFUD](..\components\sfud\src\sfud.c:727) Flash device reset success.
[SFUD]GD25Q32CSIG flash device is initialize success.
[Flash](..\..\..\..\..\easyflash\src\ef_env.c:1776) ENV start address is 0x00000000, size is 8192 bytes.

[Flash]EasyFlash V4.0.99 is initialize success.

[Flash]You can get the latest version on https://github.com/armink/EasyFlash .

assert failed at ..\APP\src\app.c:36 
读取boot_times失败。
平台：STM32F103，flash:GD25Q32CSIG 
其他配置信息均没改变。

armink, 2019-11-19:
试着先设置下 boot_times 再去读取

yeweizhong, 2019-11-19:
使用ef_set_env_blob设置后可正常读出，试了简单的读修写，都可以。谢谢作者的开源作品。

closed 2019-11-19 by yeweizhong