armink/EasyFlash #81: 写入粒度设置为64时不起作用

CkovMk, 2019-11-21:
我将 `EF_WRITE_GRAN` 定义为64，在ef_port的读写函数中打印每次读些的地址和大小，发现并没有按8字节粒度写入。
开始部分打印信息为：
```
ENV start address is 0x00000000, size is 65536 bytes.
Verbose: Read Addr 0x00000000, Size 0060 bytes
Warning: Sector header check failed. Format this sector (0x00000000).
Verbose: Erase Addr 0x00000000, Size 4096 bytes
Verbose: Write Addr 0x00000000, Size 0060 bytes
Warning: Write Failed !
Verbose: Read Addr 0x00000000, Size 0060 bytes
```
其中第五行产生了60字节写入，不符合8字节对齐。

我使用的平台是NXP的K66单片机，该单片机具有256(sectors) * 4096KB = 1MB的内部flash，写入必须8字节对齐，读取可以4字节对齐。由于我的底层驱动内部已进行过地址转换，所以暴漏给easyFlash的首地址就是0x0。

请问这可能是什么问题？谢谢。

CkovMk, 2019-11-21:
另外在 `ef_load_env(void)` 函数的 `goto __retry;` 语句附近，似乎容易进入死循环，不停地读上面地那60个字节。

armink, 2019-11-24:
试试对 `struct sector_hdr_data` 增加个保留属性，保证 64bit 对比

以前测试驱动没有你的这个严格，上层传入的不是 64bit 长度对齐，会自动在后面写入 FF ，所以没测出问题来

你先试试这样的改法吧，等你结果

CkovMk, 2019-11-28:
修改之后可以正常初始化了。
我是用的是GCC，这是修改后的代码：
```c
__attribute__((aligned(EF_WRITE_GRAN/8)))
struct sector_hdr_data {
    ...
};
```
其他编译器的支持可能还需要额外处理一下。
以下是部分串口输出结果：
```
ENV start address is 0x00000000, size is 65536 bytes.
Verbose: Read Addr 0x00000000, Size 0064 bytes
Warning: Sector header check failed. Format this sector (0x00000000).
Verbose: Erase Addr 0x00000000, Size 4096 bytes
Verbose: Write Addr 0x00000000, Size 0064 bytes
Verbose: Read Addr 0x00000000, Size 0064 bytes
Verbose: Read Addr 0x00001000, Size 0064 bytes
...
```
我这边暂时还没有测试其他写入是否正常，请问还有没有其他需要设置对齐的变量？

另外（感觉不值得另外开一个Issue）我是用的是C++编译器，直接编译的话有些类型检查不能通过，主要是C++不允许 `void*` 隐式转换为其他指针。这里我加上了几个指针的类型转换才通过编译。这点建议调整一下。

非常感谢。

armink, 2019-11-28:
`ef_port_write` 的断言先不加了吧，有些产品上的确实也支持非对齐 size 的写入功能，非对齐的部分会自动补全 FF 。

如果把 `void*` 改为 `uint8_t *` 这样的方式，可能也会不方便 C 的工程师使用，你觉得呢

CkovMk, 2019-11-28:
这样也可以吧，就是用C++时需要自己debug一下，加几个强制类型转换，影响也不大。

closed 2019-12-08 by CkovMk

augussong, 2020-10-10:
@CkovMk你好！我也遇到写入粒度设置为64时不起作用的问题， 能把ef_port.c移植的代码贴出来吗？或把移植的代码发的我邮箱caihongsw@126.com谢谢

CkovMk, 2020-10-11:
@augussong 写入粒度问题的解决方法是修改`easyflash/src/ef_env.c`。在该文件内搜索`struct sector_hdr_data`，找到定义后对其增加对齐指令即可。详细过程在上面。我对`ef_port.c`的修改是增加了对写入粒度的检测，仅仅是为了验证我修改的效果。

liudongpeng, 2024-08-21:
我写入颗粒度设置为64，初始化成功，ef_set_and_save_env()返回值是0，应该也是调用成功了，但是数据并没有到存到flash。
我按照前辈的方式，修改struct __attribute__((aligned(8))) sector_hdr_data结构体为8字节对齐，但还是不起作用。

ret = easyflash_init();
    if (ret == 0)
    {
        printf("easyflash_init() ok\n");

        ef_print_env();
    }

    ret = ef_set_and_save_env("name", "summer");
    printf("ef_set_and_save_env() ret = %d\n", ret);
    printf("{key: name, value: %s}\n", ef_get_env("name"));

[image]