armink/EasyFlash #82: 关于写入env速度问题

yeweizhong, 2019-11-26:
hi，armink。目前我使用的场景是为了保存名单列表，当我测试保存2000个名单时，需要花费30分钟才能完成。请问下有什么办法可以优化速度吗？我目前已经把crc计算给去掉了，提升并不大。而且，越到后面，写入所需时间越长。

以下是日志：
SFUD](..\components\sfud\src\sfud.c:104) Start initialize Serial Flash Universal Driver(SFUD) V0.10.29.
[SFUD](..\components\sfud\src\sfud.c:105) You can get the latest version on https://github.com/armink/SFUD .
[SFUD](..\components\sfud\src\sfud.c:748) The flash device manufacturer ID is 0xC8, memory type ID is 0x40, capacity ID is 0x16.
[SFUD]Find a GigaDevice GD25Q32CSIG flash chip. Size is 4194304 bytes.
[SFUD](..\components\sfud\src\sfud.c:727) Flash device reset success.
[SFUD]GD25Q32CSIG flash device is initialize success.
[Flash]Warning: Sector header check failed. Format this sector (0x002a9000).

[Flash]EasyFlash V4.0.99 is initialize success.

[Flash]You can get the latest version on https://github.com/armink/EasyFlash .

test_set i = 0  ---- 184Ms 
test_set i = 1  ---- 183Ms 
test_set i = 2  ---- 184Ms 
test_set i = 3  ---- 184Ms 
test_set i = 4  ---- 185Ms 
test_set i = 5  ---- 186Ms 
test_set i = 6  ---- 186Ms 
test_set i = 7  ---- 187Ms 
test_set i = 8  ---- 187Ms 
test_set i = 9  ---- 188Ms 
test_set i = 10  ---- 189Ms 
test_set i = 11  ---- 189Ms 
test_set i = 12  ---- 189Ms 
test_set i = 13  ---- 190Ms 
test_set i = 14  ---- 191Ms 
test_set i = 15  ---- 191Ms 
test_set i = 16  ---- 192Ms 
test_set i = 17  ---- 193Ms 
test_set i = 18  ---- 193Ms 
test_set i = 19  ---- 193Ms 
test_set i = 20  ---- 194Ms 
test_set i = 21  ---- 195Ms 
test_set i = 22  ---- 195Ms 
test_set i = 23  ---- 196Ms 
test_set i = 24  ---- 197Ms 
test_set i = 25  ---- 197Ms 
test_set i = 26  ---- 197Ms 
test_set i = 27  ---- 198Ms 
test_set i = 28  ---- 199Ms 
test_set i = 29  ---- 199Ms 
test_set i = 30  ---- 200Ms 
test_set i = 31  ---- 201Ms 
test_set i = 32  ---- 201Ms 
test_set i = 33  ---- 201Ms 
test_set i = 34  ---- 202Ms 
test_set i = 35  ---- 203Ms 
test_set i = 36  ---- 203Ms 
test_set i = 37  ---- 204Ms 
test_set i = 38  ---- 204Ms 
test_set i = 39  ---- 205Ms 
test_set i = 40  ---- 206Ms 
test_set i = 41  ---- 206Ms 
test_set i = 42  ---- 206Ms 
test_set i = 43  ---- 208Ms 
test_set i = 44  ---- 207Ms 
test_set i = 45  ---- 209Ms 
test_set i = 46  ---- 209Ms 
test_set i = 47  ---- 209Ms 
test_set i = 48  ---- 210Ms 
test_set i = 49  ---- 211Ms 
test_set i = 50  ---- 211Ms 
test_set i = 51  ---- 211Ms 
test_set i = 52  ---- 213Ms 
test_set i = 53  ---- 213Ms 
test_set i = 54  ---- 213Ms 
[... 1920 lines cut ...]
test_set i = 1975  ---- 1304Ms 
test_set i = 1976  ---- 1305Ms 
test_set i = 1977  ---- 1306Ms 
test_set i = 1978  ---- 1306Ms 
test_set i = 1979  ---- 1397Ms 
test_set i = 1980  ---- 1308Ms 
test_set i = 1981  ---- 1308Ms 
test_set i = 1982  ---- 1308Ms 
test_set i = 1983  ---- 1309Ms 
test_set i = 1984  ---- 1310Ms 
test_set i = 1985  ---- 1310Ms 
test_set i = 1986  ---- 1311Ms 
test_set i = 1987  ---- 1311Ms 
test_set i = 1988  ---- 1312Ms 
test_set i = 1989  ---- 1313Ms 
test_set i = 1990  ---- 1313Ms 
test_set i = 1991  ---- 1313Ms 
test_set i = 1992  ---- 1314Ms 
test_set i = 1993  ---- 1315Ms 
test_set i = 1994  ---- 1315Ms 
test_set i = 1995  ---- 1316Ms 
test_set i = 1996  ---- 1317Ms 
test_set i = 1997  ---- 1317Ms 
test_set i = 1998  ---- 1317Ms 
test_set i = 1999  ---- 1318Ms

armink, 2019-11-27:
```shell
msh />ef_bench_test 2000
01-01 00:00:18.813 D/app tshell: write target0 18 ms
01-01 00:00:18.839 D/app tshell: write target1 19 ms
01-01 00:00:18.865 D/app tshell: write target2 19 ms
01-01 00:00:18.892 D/app tshell: write target3 20 ms
01-01 00:00:18.919 D/app tshell: write target4 20 ms
01-01 00:00:18.946 D/app tshell: write target5 20 ms
01-01 00:00:18.974 D/app tshell: write target6 21 ms
01-01 00:00:18.002 D/app tshell: write target7 21 ms
01-01 00:00:19.031 D/app tshell: write target8 22 ms
01-01 00:00:19.060 D/app tshell: write target9 22 ms
01-01 00:00:19.089 D/app tshell: write target10 22 ms
01-01 00:00:19.121 D/app tshell: write target11 24 ms
01-01 00:00:19.151 D/app tshell: write target12 23 ms
01-01 00:00:19.182 D/app tshell: write target13 24 ms
01-01 00:00:19.213 D/app tshell: write target14 24 ms
01-01 00:00:19.244 D/app tshell: write target15 24 ms
01-01 00:00:19.277 D/app tshell: write target16 25 ms
01-01 00:00:19.309 D/app tshell: write target17 25 ms
01-01 00:00:19.343 D/app tshell: write target18 27 ms
01-01 00:00:19.376 D/app tshell: write target19 26 ms
01-01 00:00:19.410 D/app tshell: write target20 26 ms
01-01 00:00:19.444 D/app tshell: write target21 27 ms
01-01 00:00:19.478 D/app tshell: write target22 27 ms
01-01 00:00:19.513 D/app tshell: write target23 28 ms
01-01 00:00:19.548 D/app tshell: write target24 28 ms
01-01 00:00:19.586 D/app tshell: write target25 30 ms
01-01 00:00:19.622 D/app tshell: write target26 29 ms
01-01 00:00:19.658 D/app tshell: write target27 29 ms
01-01 00:00:19.695 D/app tshell: write target28 30 ms
01-01 00:00:19.733 D/app tshell: write target29 30 ms
01-01 00:00:19.770 D/app tshell: write target30 30 ms
01-01 00:00:19.808 D/app tshell: write target31 31 ms
01-01 00:00:19.847 D/app tshell: write target32 32 ms
01-01 00:00:19.886 D/app tshell: write target33 32 ms
01-01 00:00:19.926 D/app tshell: write target34 32 ms
01-01 00:00:19.965 D/app tshell: write target35 32 ms
01-01 00:00:20.005 D/app tshell: write target36 33 ms
01-01 00:00:20.045 D/app tshell: write target37 33 ms
01-01 00:00:20.086 D/app tshell: write target38 34 ms
01-01 00:00:20.128 D/app tshell: write target39 35 ms
01-01 00:00:20.170 D/app tshell: write target40 35 ms
01-01 00:00:20.212 D/app tshell: write target41 35 ms
01-01 00:00:20.254 D/app tshell: write target42 35 ms
01-01 00:00:20.298 D/app tshell: write target43 36 ms
01-01 00:00:20.341 D/app tshell: write target44 36 ms
01-01 00:00:20.385 D/app tshell: write target45 37 ms
01-01 00:00:20.430 D/app tshell: write target46 38 ms
01-01 00:00:20.474 D/app tshell: write target47 37 ms
01-01 00:00:20.520 D/app tshell: write target48 38 ms
01-01 00:00:20.565 D/app tshell: write target49 38 ms
01-01 00:00:20.611 D/app tshell: write target50 39 ms
01-01 00:00:20.657 D/app tshell: write target51 39 ms
01-01 00:00:20.704 D/app tshell: write target52 39 ms
01-01 00:00:20.752 D/app tshell: write target53 41 ms
01-01 00:00:20.799 D/app tshell: write target54 40 ms
01-01 00:00:20.847 D/app tshell: write target55 41 ms
01-01 00:00:20.895 D/app tshell: write target56 41 ms
01-01 00:00:20.999 D/app tshell: write target57 42 ms
01-01 00:00:21.048 D/app tshell: write target58 42 ms
01-01 00:00:21.097 D/app tshell: write target59 42 ms
01-01 00:00:21.147 D/app tshell: write target60 43 ms
01-01 00:00:21.198 D/app tshell: write target61 43 ms
01-01 00:00:21.249 D/app tshell: write target62 44 ms
01-01 00:00:21.300 D/app tshell: write target63 44 ms
01-01 00:00:21.351 D/app tshell: write target64 44 ms
01-01 00:00:21.403 D/app tshell: write target65 45 ms
01-01 00:00:21.456 D/app tshell: write target66 45 ms
01-01 00:00:21.509 D/app tshell: write target67 46 ms
[... 1950 lines cut ...]
01-01 00:14:52.610 D/app tshell: write target1982 850 ms
01-01 00:14:53.468 D/app tshell: write target1983 851 ms
01-01 00:14:54.326 D/app tshell: write target1984 851 ms
01-01 00:14:55.185 D/app tshell: write target1985 851 ms
01-01 00:14:56.044 D/app tshell: write target1986 852 ms
01-01 00:14:56.903 D/app tshell: write target1987 852 ms
01-01 00:14:57.762 D/app tshell: write target1988 852 ms
01-01 00:14:58.624 D/app tshell: write target1989 854 ms
01-01 00:14:59.484 D/app tshell: write target1990 853 ms
01-01 00:15:00.345 D/app tshell: write target1991 854 ms
01-01 00:15:01.206 D/app tshell: write target1992 854 ms
01-01 00:15:02.069 D/app tshell: write target1993 855 ms
01-01 00:15:02.931 D/app tshell: write target1994 855 ms
01-01 00:15:03.793 D/app tshell: write target1995 855 ms
01-01 00:15:04.657 D/app tshell: write target1996 857 ms
01-01 00:15:05.520 D/app tshell: write target1997 856 ms
01-01 00:15:06.384 D/app tshell: write target1998 857 ms
01-01 00:15:07.248 D/app tshell: write target1999 857 ms
```

我在自己的板子上测试了，速度应该比你的快一倍，但也还是挺慢的

数据量大以后，时间主要花在了 **查重** 上面，避免有重复 key name 的情况

你那边主要的需求是什么样的？kv 最多多少？是否存在修改 kv 的情况？

yeweizhong, 2019-11-27:
主要保存名单信息，kv数据可能要达到10000+，有修改的需求。所以easyflash可能不适用这种大数据的场景。

armink, 2019-11-27:
可以换个角度想，其实 10000+ 个应该不会同一时间一起加入吧？

如果不会，本身 EasyFlash 还有 Cache 机制，到很多个 kv 以后，虽然第一次写入可能慢一些，但第二次访问 kv 会非常的快

现在 EasyFlash 内存消耗非常小，所以提升大数据查询效率还是有点难度。不知道你对内存要求高不高，如果内存充足，其实还是有很多空间换时间的办法

armink, 2019-11-27:
另外，EasyFlash 在 V5.0 会支持多实例，此时 10000+ 的 kv 可以分类索引到不同的 EasyFlash 对象区域内，对大数据量下，提升效率应该还是有帮助的

yeweizhong, 2019-11-28:
根据大神的代码逻辑，简化了下流程，自己弄个简单的flash读写库，现在写10000+数据，只需要用2000ms了。谢谢！

closed 2019-11-28 by yeweizhong

armink, 2019-11-28:
具体怎么操作的呢，可以跟大家分享下嘛

yeweizhong, 2019-11-28:
因为我的名单数据来源都是通过后台传输的，所以可以下载全量数据，然后名单会标识名单是否新增和修改或者删除，根据这个标识减少查重的次数。然后flash主要分2个大扇区来存放数据（大约5W个名单），通过标头去记录写入了那个扇区，和写入的扇区的地址。每次名单数据下载完就会判断扇区是否写满，当扇区写满了，会触发告诉后台下次需要下载全量数据，写入另外一个扇区。因为前一个扇区也是会存在删除的数据，就是脏数据。就是一个很简单的双扇区循环读写。