armink/EasyFlash #86: 对Env的Key不断地删除、写入数据，一段时间后，sector满了，未对sector进行初始化，导致Env不能在写入

KamKeung, 2019-12-25:
对Env的Key不断地删除、写入数据，一段时间后，会进入到EF_DEBUG("Trigger a GC check after created ENV.\n"); 
而sector->check_ok = 1,
    sector->status.store = SECTOR_STORE_USING, 
    sector->status.dirty = SECTOR_DIRTY_FALSE。
导致在do_gc()函数中未进入if (sector->check_ok && (sector->status.dirty == SECTOR_DIRTY_TRUE || sector->status.dirty == SECTOR_DIRTY_GC)) 失败，而未启动初始化。
请问：是什么原因或某些配置未配置好导致 sector->status.dirty 一直为SECTOR_DIRTY_FALSE，不为SECTOR_DIRTY_TRUE或SECTOR_DIRTY_GC。

armink, 2019-12-25:
`SECTOR_DIRTY_FALSE` 说明确实不是脏的，也是没有存在被删除或者修改的环境变量

KamKeung, 2019-12-26:
对Env的同一Key不断地写入数据(不在应用层删除，set_env函数中有检测相同Key会将old Key删除，此时 Sector->status.dirty是不是应该被设为SECTOR_DIRTY_TRUE？但是查看全局变量sector，sector->status.dirty = SECTOR_DIRTY_FALSE ？)，写入数据的起始地址 不断移动，直到 移到到 EF_ERASE_MIN_SIZE，启动GC，但是此时的sector->status.dirty = SECTOR_DIRTY_FALSE无法进行Move_Env和格式化。导致 写入数据的起始地址为 ENV_AREA_SIZE时， 无法在写入数据。
请问： 怎样的操作才能将 sector->status.dirty 设为 SECTOR_DIRTY_TRUE？

KamKeung, 2019-12-27:
我在ef_env.c文件中找到唯一能设SECTOR_DIRTY_TRUE 是在del_env()函数中**result = write_status(dirty_status_addr, status_table, SECTOR_DIRTY_STATUS_NUM, SECTOR_DIRTY_TRUE);**但这是对env中的old_Key的status.dirty进行设值，而不是对sector->status.dirty进行设值。
而唯一对sector->status.dirty进行设值 是在read_sector_meta_data()函数中的get_status()函数，但是get_status()获取的状态是地址sector_cache_table[x].addr中的status.dirty，而不是old_Key的status.dirty。

zh0nglihua, 2023-07-29:
这个问题最后解决了吗，我现在在4.0也遇到了。

zh0nglihua, 2023-07-29:
@KamKeung