armink/EasyFlash #91: 查找不存在 Key 的效率需要想办法提升下，有什么思路讨论下

wujun8, 2020-01-10:
当 Key 不存在的情况下，现有的逻辑是要遍历所有扇区的每一个 ENV 才能确定是不存在的

wujun8, 2020-01-10:
查找一个 Key 还对写入造成很大的影响，因为每次写入前都要查找是否有 old_env ，对于这一点，建议开放一个接口只写入新 Key 的，业务上已经保证这个 Key 是不存在的，就没必要再去遍历了，但是这个如果被误用也不好

armink, 2020-01-10:
是的，如果定期 GC 可能也可以缓解下这个问题

wujun8, 2020-01-13:
定期 GC 只能减少脏数据的遍历时间，且耗时较长，触发点放在什么位置也不好选择

armink, 2020-01-13:
或者根据 key 类别进行分区存储

wujun8, 2020-01-16:
根据 key 类别进行分区可以通过多实例来做；
是不是可以这样，按照 key 计算 hash 分布到哪个扇区，冲突的后移查找可用扇区，扇区内读写模式保持原有设计，这样理想的情况下，查找一个 key 只需遍历一个扇区的数据，这里的关键点是怎么构建一个冲突率最低的 hash 函数

wujun8, 2020-01-17:
这样 hash 取值范围太小了，不管是用扇区编号还是划分更小的区块，都很容易冲突

armink, 2020-01-19:
是的，所以是否可以在用户那边做好分类（类似 tag ），ENV 使用的时候，增加类别参数

yc-2503, 2020-02-12:
> 根据 key 类别进行分区可以通过多实例来做；
> 是不是可以这样，按照 key 计算 hash 分布到哪个扇区，冲突的后移查找可用扇区，扇区内读写模式保持原有设计，这样理想的情况下，查找一个 key 只需遍历一个扇区的数据，这里的关键点是怎么构建一个冲突率最低的 hash 函数

------------------------------------------------------------------
初始化的时候，做张表，对KEY进行排序，标记好KEY对应的扇区和地址，是不是好搞一些。
另外我觉得EasyFlash主要是提供KeyValue的底层接口，应该足够精简高效，像这种提高查找效率的问题应该由上一层处理，根据具体的应用场景，比如数据量的大小、是否用操作系统、是否需要排序、是否需要按条件查询等，具体实现。

armink, 2020-05-06:
现在支持多实例了，可以试试新版本

CkovMk, 2020-11-11:
“多实例”具体如何使用呢？我的应用刚好需要管理片内Flash和外部EEPROM，想了解一下。

armink, 2020-11-11:
> “多实例”具体如何使用呢？我的应用刚好需要管理片内Flash和外部EEPROM，想了解一下。

目前 FlashDB 已经支持多实例，可以试试

wu1045718093, 2021-06-30:
上层加个布隆过滤器即可