armink/EasyFlash #100: EasyFlash无法正常使用的问题

xiaodingding, 2020-05-12:
1.芯片：STM32L151CBT6。
2.报错函数：
void ef_port_env_lock(void) {
    rt_sem_take(&env_cache_lock, RT_WAITING_FOREVER);
}
3.分区名称 “ef”
4.fal接口
const struct fal_flash_dev eeprom0 = { "eeprom0", 0, 16*1024, 64, {init, read, write, erase} };
5.分区表
#define FAL_PART_TABLE                                                      
{                                                                                
    {FAL_PART_MAGIC_WROD, "ef", "eeprom0",   0,    12 * 1024,  0}, 
}
6.fal组件正常，能够正常读写外置flash的数据，但是EasyFlash，一直不能正常工作。
7.EasyFlash日志
 0> at24cxx_init for fal init successful 
 0> [D/FAL] (fal_flash_init:61) Flash device |                  eeprom0 | addr: 0x00000000 | len: 0x00004000 | blk_size: 0x00000040 |initialized finish.
 0> [I/FAL] ==================== FAL partition table ====================
 0> [I/FAL] | name | flash_dev |   offset   |    length  |
 0> [I/FAL] -------------------------------------------------------------
 0> [I/FAL] | ef   | eeprom0   | 0x00000000 | 0x00003000 |
 0> [I/FAL] =============================================================
 0> [I/FAL] RT-Thread Flash Abstraction Layer (V0.2.0) initialize success.
 0> msh >spi init successful, find spi10 device! 
 0> result:3, hu_data:1f4 
 0> [E/DBG] Failed write PWR_REG, not support power manager
 0> [E/DBG] Failed write PWR_REG, not support power manager
 0> init message queue failed.
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1818) ENV start address is 0x00000000, size is 8192 bytes.
 0> at24cxx_page_read offset:0 size:16 
 0> [Flash] Warning: Sector header check failed. Format this sector (0x000at24cxx earse is null function 
 0> moto_thread_entry start!
 0> at24cxx_page_write offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] Warning: Sector header check failed. Format this sector (0x000at24cxx earse is null function 
 0> at24cxx_page_write offset:4096 size:16 
 0> [Flash] Warning: All sector header check failed. Set it to default.
 0> at24cxx earse is null function 
 0> at24cxx_page_write offset:0 size:16 
 0> at24cxx earse is null function 
 0> at24cxx_page_write offset:4096 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_write offset:0 size:1 
 0> at24cxx_page_write offset:16 size:1 
 0> adxl362 move interrupt 
 0> adxl362 move interrupt 
 0> at24cxx_page_write offset:20 size:20 
 0> at24cxx_page_write offset:40 size:17 
 0> at24cxx_page_write offset:57 size:1 
 0> at24cxx_page_write offset:16 size:1 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1059) Trigger a GC check after alloc ENV failed.
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1196) Warning: Alloc an ENV (size 45) failed when new ENV. Now will at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1264) The remain empty sector is 1, GC threshold is 1.
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1059) Trigger a GC check after alloc ENV failed.
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> at24cxx_page_write offset:4096 size:1 
 0> at24cxx_page_write offset:4112 size:1 
 0> at24cxx_page_write offset:4116 size:20 
 0> at24cxx_page_write offset:4136 size:17 
 0> at24cxx_page_write offset:4153 size:1 
 0> at24cxx_page_write offset:4112 size:1 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1196) Warning: Alloc an ENV (size 43) failed when new ENV. Now will at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1264) The remain empty sector is 0, GC threshold is 1.
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> (rt_object_get_type(&sem->parent.parent) == RT_Object_Class_Se

8.注释掉ef_port_env_lock和ef_port_env_unlock的内容，还是无法正常运行。

armink, 2020-05-12:
看着像内存被写穿了，试试换为 mutex 呢

xiaodingding, 2020-05-12:
是不是因为内存不够的原因

xiaodingding, 2020-05-12:
EfErrCode ef_port_init(ef_env const **default_env, size_t *default_env_size) {
    EfErrCode result = EF_NO_ERR;

    *default_env = default_env_set;
    *default_env_size = sizeof(default_env_set) / sizeof(default_env_set[0]);

    // rt_sem_init(&env_cache_lock, "env lock", 1, RT_IPC_FLAG_PRIO); 刚注释的

    env_cache_mutex = rt_mutex_create("emutex", RT_IPC_FLAG_FIFO);
    if(env_cache_mutex == RT_NULL)
    {
        rt_kprintf("create easyflash dynamic mutex failed.\n");
        return -1;
    }

    part = fal_partition_find(FAL_EF_PART_NAME);
    EF_ASSERT(part);

    return result;
}

xiaodingding, 2020-05-12:
以下是改为mutex的日志：
 0> at24cxx_init for fal init successful 
 0> [D/FAL] (fal_flash_init:61) Flash device |                  eeprom0 | addr: 0x00000000 | len: 0x00004000 | blk_size: 0x00000040 |initialized finish.
 0> [I/FAL] ==================== FAL partition table ====================
 0> [I/FAL] | name | flash_dev |   offset   |    length  |
 0> [I/FAL] -------------------------------------------------------------
 0> [I/FAL] | ef   | eeprom0   | 0x00000000 | 0x00003000 |
 0> [I/FAL] =============================================================
 0> [I/FAL] RT-Thread Flash Abstraction Layer (V0.2.0) initialize success.
 0> msh >spi init successful, find spi10 device! 
 0> result:3, hu_data:1f4 
 0> [E/DBG] Failed write PWR_REG, not support power manager
 0> [E/DBG] Failed write PWR_REG, not support power manager
 0> init message queue failed.
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1818) ENV start address is 0x00000000, size is 8192 bytes.
 0> at24cxx_page_read offset:0 size:16 
 0> [Flash] Warning: Sector header check failed. Format this sector (0x000at24cxx earse write offset:0 ,size:4096 
 0> moto_thread_entry start!
 0> at24cxx earse write addr:0 ,yushu:0, result:0 
 0> adxl362 move interrupt 
 0> adxl362 move interrupt 
 0> at24cxx earse write addr:128 ,yushu:1, result:0 
 0> at24cxx earse write addr:256 ,yushu:2, result:0 
 0> at24cxx earse write addr:384 ,yushu:3, result:0 
 0> at24cxx earse write addr:512 ,yushu:4, result:0 
 0> at24cxx earse write addr:640 ,yushu:5, result:0 
 0> at24cxx earse write addr:768 ,yushu:6, result:0 
 0> at24cxx earse write addr:896 ,yushu:7, result:0 
 0> at24cxx earse write addr:1024 ,yushu:8, result:0 
 0> at24cxx earse write addr:1152 ,yushu:9, result:0 
 0> at24cxx earse write addr:1280 ,yushu:10, result:0 
 0> at24cxx earse write addr:1408 ,yushu:11, result:0 
 0> at24cxx earse write addr:1536 ,yushu:12, result:0 
 0> at24cxx earse write addr:1664 ,yushu:13, result:0 
 0> at24cxx earse write addr:1792 ,yushu:14, result:0 
 0> at24cxx earse write addr:1920 ,yushu:15, result:0 
 0> at24cxx earse write addr:2048 ,yushu:16, result:0 
 0> at24cxx earse write addr:2176 ,yushu:17, result:0 
 0> at24cxx earse write addr:2304 ,yushu:18, result:0 
 0> at24cxx earse write addr:2432 ,yushu:19, result:0 
 0> at24cxx earse write addr:2560 ,yushu:20, result:0 
 0> at24cxx earse write addr:2688 ,yushu:21, result:0 
 0> at24cxx earse write addr:2816 ,yushu:22, result:0 
 0> at24cxx earse write addr:2944 ,yushu:23, result:0 
 0> at24cxx earse write addr:3072 ,yushu:24, result:0 
 0> at24cxx earse write addr:3200 ,yushu:25, result:0 
 0> at24cxx earse write addr:3328 ,yushu:26, result:0 
 0> at24cxx earse write addr:3456 ,yushu:27, result:0 
 0> at24cxx earse write addr:3584 ,yushu:28, result:0 
 0> at24cxx earse write addr:3712 ,yushu:29, result:0 
 0> at24cxx earse write addr:3840 ,yushu:30, result:0 
 0> at24cxx earse write addr:3968 ,yushu:31, result:0 
 0> at24cxx earse  offset:0 ,size:4096 ,yushu:0 ,loop:32  
 0> at24cxx earse at24cxx_page_write to earse size:4096 
 0> at24cxx earse at24cxx_page_write to earse size:4096 offset:0
 0> at24cxx_page_write offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] Warning: Sector header check failed. Format this sector (0x000at24cxx earse write offset:4096 ,size:4096 
 0> at24cxx earse write addr:4096 ,yushu:0, result:0 
 0> at24cxx earse write addr:4224 ,yushu:1, result:0 
 0> at24cxx earse write addr:4352 ,yushu:2, result:0 
 0> at24cxx earse write addr:4480 ,yushu:3, result:0 
 0> at24cxx earse write addr:4608 ,yushu:4, result:0 
 0> at24cxx earse write addr:4736 ,yushu:5, result:0 
 0> at24cxx earse write addr:4864 ,yushu:6, result:0 
 0> at24cxx earse write addr:4992 ,yushu:7, result:0 
 0> at24cxx earse write addr:5120 ,yushu:8, result:0 
 0> at24cxx earse write addr:5248 ,yushu:9, result:0 
 0> at24cxx earse write addr:5376 ,yushu:10, result:0 
 0> at24cxx earse write addr:5504 ,yushu:11, result:0 
[... 125 lines cut ...]
 0> at24cxx_page_write offset:4112 size:1 
 0> at24cxx_page_write offset:4116 size:20 
 0> at24cxx_page_write offset:4136 size:17 
 0> at24cxx_page_write offset:4153 size:1 
 0> at24cxx_page_write offset:4112 size:1 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1196) Warning: Alloc an ENV (size 43) failed when new ENV. Now will at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1264) The remain empty sector is 0, GC threshold is 1.
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> [Flash] EasyFlash V4.1.0 is initialize success.
 0> [Flash] You can get the latest version on https://github.com/armink/Eaeasy falsh init successful 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16 
 0> The system now boot 1 times 
 0> at24cxx_page_read offset:0 size:16 
 0> at24cxx_page_read offset:4096 size:16

xiaodingding, 2020-05-12:
接口实现：
/*
 * This file is part of the EasyFlash Library.
 *
 * Copyright (c) 2015, Armink, <armink.ztl@gmail.com>
 *
 * Permission is hereby granted, free of charge, to any person obtaining
 * a copy of this software and associated documentation files (the
 * 'Software'), to deal in the Software without restriction, including
 * without limitation the rights to use, copy, modify, merge, publish,
 * distribute, sublicense, and/or sell copies of the Software, and to
 * permit persons to whom the Software is furnished to do so, subject to
 * the following conditions:
 *
 * The above copyright notice and this permission notice shall be
 * included in all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED 'AS IS', WITHOUT WARRANTY OF ANY KIND,
 * EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF
 * MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.
 * IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY
 * CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT,
 * TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE
 * SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 * Function: Portable interface for FAL (Flash Abstraction Layer) partition.
 * Created on: 2018-05-19
 */

#include <easyflash.h>
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <rthw.h>
#include <rtthread.h>
#include <fal.h>

/* EasyFlash partition name on FAL partition table */
#define FAL_EF_PART_NAME               "ef"

#define  RT_CONSOLEBUF_SIZE 64

/* default ENV set for user */
static const ef_env default_env_set[] = {
        {"iap_need_copy_app", "0"},
        {"iap_need_crc32_check", "0"},
        {"iap_copy_app_size", "0"},
        {"stop_in_bootloader", "0"},
};

static char log_buf[RT_CONSOLEBUF_SIZE];
// static struct rt_semaphore env_cache_lock;
static rt_mutex_t env_cache_mutex = RT_NULL;
static const struct fal_partition *part = NULL;

/**
 * Flash port for hardware initialize.
 *
 * @param default_env default ENV set for user
 * @param default_env_size default ENV size
 *
 * @return result
 */
EfErrCode ef_port_init(ef_env const **default_env, size_t *default_env_size) {
    EfErrCode result = EF_NO_ERR;

    *default_env = default_env_set;
    *default_env_size = sizeof(default_env_set) / sizeof(default_env_set[0]);

    // rt_sem_init(&env_cache_lock, "env lock", 1, RT_IPC_FLAG_PRIO);

    env_cache_mutex = rt_mutex_create("emutex", RT_IPC_FLAG_FIFO);
    if(env_cache_mutex == RT_NULL)
    {
        rt_kprintf("create easyflash dynamic mutex failed.\n");
        return -1;
    }

    part = fal_partition_find(FAL_EF_PART_NAME);
    EF_ASSERT(part);

    return result;
}

/**
 * Read data from flash.
 * @note This operation's units is word.
 *
 * @param addr flash address
 * @param buf buffer to store read data
 * @param size read bytes size
 *
 * @return result
 */
EfErrCode ef_port_read(uint32_t addr, uint32_t *buf, size_t size) {
    EfErrCode result = EF_NO_ERR;

    fal_partition_read(part, addr, (uint8_t *)buf, size);

    return result;
}

/**
 * Erase data on flash.
 * @note This operation is irreversible.
 * @note This operation's units is different which on many chips.
 *
 * @param addr flash address
 * @param size erase bytes size
 *
 * @return result
 */
EfErrCode ef_port_erase(uint32_t addr, size_t size) {
    EfErrCode result = EF_NO_ERR;

    /* make sure the start address is a multiple of FLASH_ERASE_MIN_SIZE */
    EF_ASSERT(addr % EF_ERASE_MIN_SIZE == 0);

    if (fal_partition_erase(part, addr, size) < 0)
    {
        result = EF_ERASE_ERR;
    }

    return result;
}
/**
 * Write data to flash.
 * @note This operation's units is word.
 * @note This operation must after erase. @see flash_erase.
 *
 * @param addr flash address
 * @param buf the write data buffer
 * @param size write bytes size
 *
 * @return result
 */
EfErrCode ef_port_write(uint32_t addr, const uint32_t *buf, size_t size) {
    EfErrCode result = EF_NO_ERR;

    if (fal_partition_write(part, addr, (uint8_t *)buf, size) < 0)
    {
        result = EF_WRITE_ERR;
    }

    return result;
}

/**
 * lock the ENV ram cache
 */
void ef_port_env_lock(void) {
    // rt_sem_take(&env_cache_lock, RT_WAITING_FOREVER);
    rt_mutex_take(env_cache_mutex, RT_WAITING_FOREVER);
}

/**
 * unlock the ENV ram cache
 */
void ef_port_env_unlock(void) {
    // rt_sem_release(&env_cache_lock);
    rt_mutex_release(env_cache_mutex);
}

/**
 * This function is print flash debug info.
 *
 * @param file the file which has call this function
 * @param line the line number which has call this function
 * @param format output format
 * @param ... args
 *
 */
void ef_log_debug(const char *file, const long line, const char *format, ...) {

#ifdef PRINT_DEBUG

    va_list args;

    /* args point to the first variable parameter */
    va_start(args, format);
    ef_print("[Flash] (%s:%ld) ", file, line);
    /* must use vprintf to print */
    rt_vsprintf(log_buf, format, args);
    ef_print("%s", log_buf);
    va_end(args);

#endif

}

/**
 * This function is print flash routine info.
 *
 * @param format output format
 * @param ... args
 */
void ef_log_info(const char *format, ...) {
    va_list args;

    /* args point to the first variable parameter */
    va_start(args, format);
    ef_print("[Flash] ");
    /* must use vprintf to print */
    rt_vsprintf(log_buf, format, args);
    ef_print("%s", log_buf);
    va_end(args);
}
/**
 * This function is print flash non-package info.
 *
 * @param format output format
 * @param ... args
 */
void ef_print(const char *format, ...) {
    va_list args;

    /* args point to the first variable parameter */
    va_start(args, format);
    /* must use vprintf to print */
    rt_vsprintf(log_buf, format, args);
    rt_kprintf("%s", log_buf);
    va_end(args);
}

armink, 2020-05-12:
增大下初始化线程的堆栈试试

xiaodingding, 2020-05-13:
目前已经将main线程的堆栈改为4096了，之前的空间大小为1024,现在能看到提示操作成功，但是重启之后，它还是会重新去操作一遍flash。

0> at24cxx_init for fal init successful 
 0> [D/FAL] (fal_flash_init:61) Flash device |                  eeprom0 | addr: 0x00000000 | len: 0x00004000 | blk_size: 0x00000040 |initialized finish.
 0> [I/FAL] ==================== FAL partition table ====================
 0> [I/FAL] | name | flash_dev |   offset   |    length  |
 0> [I/FAL] -------------------------------at24cxx earse write addr:1152 ,yushu:1, result:0 
 0> at24cxx earse write addr:1280 ,yushu:2, result:0 
 0> at24cxx earse write addr:1408 ,yushu:3, result:0 
 0> at24cxx earse write addr:1536 ,yushu:4, result:0 
 0> at24cxx earse write addr:1664 ,yushu:5, result:0 
 0> at24cxx earse write addr:1792 ,yushu:6, result:0 
 0> at24cxx earse write addr:1920 ,yushu:7, result:0 
 0> at24cxx earse  offset:1024 ,size:1024 ,yushu:0 ,loop:8  
 0> at24cxx earse at24cxx_page_write to earse size:1024 
 0> at24cxx earse at24cxx_page_write to earse size:1024 offset:102[Flash] Warning: All sector header check failed. Set it to default.
 0> at24cxx earse write offset:0 ,size:1024 
 0> at24cxx earse write addr:0 ,yushu:0, result:0 
 0> at24cxx earse write addr:128 ,yushu:1, result:0 
 0> at24cxx earse write addr:256 ,yushu:2, result:0 
 0> at24cxx earse write addr:384 ,yushu:3, result:0 
 0> at24cxx earse write addr:512 ,yushu:4, result:0 
 0> at24cxx earse write addr:640 ,yushu:5, result:0 
 0> at24cxx earse write addr:768 ,yushu:6, result:0 
 0> at24cxx earse write addr:896 ,yushu:7, result:0 
 0> at24cxx earse  offset:0 ,size:1024 ,yushu:0 ,loop:8  
 0> at24cxx earse at24cxx_page_write to earse size:1024 
 0> at24cxx earse at24cxx_page_write to earse size:1024 offset:0
 0> at24cxx earse write offset:1024 ,size:1024 
 0> at24cxx earse write addr:1024 ,yushu:0, result:0 
 0> at24cxx earse write addr:1152 ,yushu:1, result:0 
 0> at24cxx earse write addr:1280 ,yushu:2, result:0 
 0> at24cxx earse write addr:1408 ,yushu:3, result:0 
 0> at24cxx earse write addr:1536 ,yushu:4, result:0 
 0> at24cxx earse write addr:1664 ,yushu:5, result:0 
 0> at24cxx earse write addr:1792 ,yushu:6, result:0 
 0> at24cxx earse write addr:1920 ,yushu:7, result:0 
 0> at24cxx earse  offset:1024 ,size:1024 ,yushu:0 ,loop:8  
 0> at24cxx earse at24cxx_page_write to earse size:1024 
 0> at24cxx earse at24cxx_page_write to earse size:1024 offset:102[Flash] EasyFlash V4.1.0 is initialize success.
 0> [Flash] You can get the latest version on https://github.com/armink/Eaeasy falsh init successful 
 0> The system now boot 1 times 
  < reboot
 0> reboot
 0> 
 0>  \ | /
 0> - RT -     Thread Operating System
 0>  / | \     4.0.3 build May 13 2020
 0>  2006 - 2020 Copyright by rt-thread team
 0> at24cxx_init for fal init successful 
 0> [D/FAL] (fal_flash_init:61) Flash device |                  eeprom0 | addr: 0x00000000 | len: 0x00004000 | blk_size: 0x00000040 |initialized finish.
 0> [I/FAL] ==================== FAL partition table ====================
 0> [I/FAL] | name | flash_dev |   offset   |    length  |
 0> [I/FAL] -------------------------------------------------------------
 0> [I/FAL] | ef   | eeprom0   | 0x00000000 | 0x00003000 |
 0> [I/FAL] =============================================================
 0> [I/FAL] RT-Thread Flash Abstraction Layer (V0.2.0) initialize success.
 0> [Flash] (packages\EasyFlash-v4.1.0\src\ef_env.c:1818) ENV start address is 0x00000000, size is 2048 bytes.
 0> msh >[Flash] Warning: Sector header check failed. Format this sector (0x000at24cxx earse write offset:0 ,size:1024 
 0> at24cxx earse write addr:0 ,yushu:0, result:0 
 0> at24cxx earse write addr:128 ,yushu:1, result:0 
 0> at24cxx earse write addr:256 ,yushu:2, result:0 
 0> at24cxx earse write addr:384 ,yushu:3, result:0 
 0> at24cxx earse write addr:512 ,yushu:4, result:0 
 0> at24cxx earse write addr:640 ,yushu:5, result:0 
 0> at24cxx earse write addr:768 ,yushu:6, result:0 
 0> at24cxx earse write addr:896 ,yushu:7, result:0 
 0> at24cxx earse  offset:0 ,size:1024 ,yushu:0 ,loop:8  
 0> at24cxx earse at24cxx_page_write to earse size:1024 
 0> at24cxx earse at24cxx_page_write to earse size:1024 offset:0
[... 13 lines cut ...]
 0> at24cxx earse write addr:0 ,yushu:0, result:0 
 0> at24cxx earse write addr:128 ,yushu:1, result:0 
 0> at24cxx earse write addr:256 ,yushu:2, result:0 
 0> at24cxx earse write addr:384 ,yushu:3, result:0 
 0> at24cxx earse write addr:512 ,yushu:4, result:0 
 0> at24cxx earse write addr:640 ,yushu:5, result:0 
 0> at24cxx earse write addr:768 ,yushu:6, result:0 
 0> at24cxx earse write addr:896 ,yushu:7, result:0 
 0> at24cxx earse  offset:0 ,size:1024 ,yushu:0 ,loop:8  
 0> at24cxx earse at24cxx_page_write to earse size:1024 
 0> at24cxx earse at24cxx_page_write to earse size:1024 offset:0
 0> at24cxx earse write offset:1024 ,size:1024 
 0> at24cxx earse write addr:1024 ,yushu:0, result:0 
 0> at24cxx earse write addr:1152 ,yushu:1, result:0 
 0> at24cxx earse write addr:1280 ,yushu:2, result:0 
 0> at24cxx earse write addr:1408 ,yushu:3, result:0 
 0> at24cxx earse write addr:1536 ,yushu:4, result:0 
 0> at24cxx earse write addr:1664 ,yushu:5, result:0 
 0> at24cxx earse write addr:1792 ,yushu:6, result:0 
 0> at24cxx earse write addr:1920 ,yushu:7, result:0 
 0> at24cxx earse  offset:1024 ,size:1024 ,yushu:0 ,loop:8  
 0> at24cxx earse at24cxx_page_write to earse size:1024 
 0> at24cxx earse at24cxx_page_write to earse size:1024 offset:102[Flash] EasyFlash V4.1.0 is initialize success.
 0> [Flash] You can get the latest version on https://github.com/armink/Eaeasy falsh init successful 
 0> The system now boot 1 times

armink, 2020-05-17:
你用的是 EEPROM 吗？

xiaodingding, 2020-05-18:
嗯，是的，是外置的eeprom

armink, 2020-05-20:
你的 ef_cfg.h 贴一些，确保写粒度有配置正确

xiaodingding, 2020-05-20:
/*
 * This file is part of the EasyFlash Library.
 *
 * Copyright (c) 2015, Armink, <armink.ztl@gmail.com>
 *
 * Permission is hereby granted, free of charge, to any person obtaining
 * a copy of this software and associated documentation files (the
 * 'Software'), to deal in the Software without restriction, including
 * without limitation the rights to use, copy, modify, merge, publish,
 * distribute, sublicense, and/or sell copies of the Software, and to
 * permit persons to whom the Software is furnished to do so, subject to
 * the following conditions:
 *
 * The above copyright notice and this permission notice shall be
 * included in all copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED 'AS IS', WITHOUT WARRANTY OF ANY KIND,
 * EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF
 * MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT.
 * IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY
 * CLAIM, DAMAGES OR OTHER LIABILITY, WHETHER IN AN ACTION OF CONTRACT,
 * TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE
 * SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
 *
 * Function: Portable interface for FAL (Flash Abstraction Layer) partition.
 * Created on: 2018-05-19
 */

#include <easyflash.h>
#include <stdio.h>
#include <stdlib.h>
#include <stdarg.h>
#include <rthw.h>
#include <rtthread.h>
#include <fal.h>

/* EasyFlash partition name on FAL partition table */
#define FAL_EF_PART_NAME               "ef"

#define  RT_CONSOLEBUF_SIZE 64

/* default ENV set for user */
static const ef_env default_env_set[] = {
        {"iap_need_copy_app", "0"},
        {"iap_need_crc32_check", "0"},
        {"iap_copy_app_size", "0"},
        {"stop_in_bootloader", "0"},
};

static char log_buf[RT_CONSOLEBUF_SIZE];
// static struct rt_semaphore env_cache_lock;
static rt_mutex_t env_cache_mutex = RT_NULL;
static const struct fal_partition *part = NULL;

/**
 * Flash port for hardware initialize.
 *
 * @param default_env default ENV set for user
 * @param default_env_size default ENV size
 *
 * @return result
 */
EfErrCode ef_port_init(ef_env const **default_env, size_t *default_env_size) {
    EfErrCode result = EF_NO_ERR;

    *default_env = default_env_set;
    *default_env_size = sizeof(default_env_set) / sizeof(default_env_set[0]);

    // rt_sem_init(&env_cache_lock, "env lock", 1, RT_IPC_FLAG_PRIO);

    env_cache_mutex = rt_mutex_create("emutex", RT_IPC_FLAG_FIFO);
    if(env_cache_mutex == RT_NULL)
    {
        rt_kprintf("create easyflash dynamic mutex failed.\n");
        return -1;
    }

    part = fal_partition_find(FAL_EF_PART_NAME);
    EF_ASSERT(part);

    return result;
}

/**
 * Read data from flash.
 * @note This operation's units is word.
 *
 * @param addr flash address
 * @param buf buffer to store read data
 * @param size read bytes size
 *
 * @return result
 */
EfErrCode ef_port_read(uint32_t addr, uint32_t *buf, size_t size) {
    EfErrCode result = EF_NO_ERR;

    fal_partition_read(part, addr, (uint8_t *)buf, size);

    return result;
}

/**
 * Erase data on flash.
 * @note This operation is irreversible.
 * @note This operation's units is different which on many chips.
 *
 * @param addr flash address
 * @param size erase bytes size
 *
 * @return result
 */
EfErrCode ef_port_erase(uint32_t addr, size_t size) {
    EfErrCode result = EF_NO_ERR;

    /* make sure the start address is a multiple of FLASH_ERASE_MIN_SIZE */
    EF_ASSERT(addr % EF_ERASE_MIN_SIZE == 0);

    if (fal_partition_erase(part, addr, size) < 0)
    {
        result = EF_ERASE_ERR;
    }

    return result;
}
/**
 * Write data to flash.
 * @note This operation's units is word.
 * @note This operation must after erase. @see flash_erase.
 *
 * @param addr flash address
 * @param buf the write data buffer
 * @param size write bytes size
 *
 * @return result
 */
EfErrCode ef_port_write(uint32_t addr, const uint32_t *buf, size_t size) {
    EfErrCode result = EF_NO_ERR;

    if (fal_partition_write(part, addr, (uint8_t *)buf, size) < 0)
    {
        result = EF_WRITE_ERR;
    }

    return result;
}

/**
 * lock the ENV ram cache
 */
void ef_port_env_lock(void) {
    // rt_sem_take(&env_cache_lock, RT_WAITING_FOREVER);
    rt_mutex_take(env_cache_mutex, RT_WAITING_FOREVER);
}

/**
 * unlock the ENV ram cache
 */
void ef_port_env_unlock(void) {
    // rt_sem_release(&env_cache_lock);
    rt_mutex_release(env_cache_mutex);
}

/**
 * This function is print flash debug info.
 *
 * @param file the file which has call this function
 * @param line the line number which has call this function
 * @param format output format
 * @param ... args
 *
 */
void ef_log_debug(const char *file, const long line, const char *format, ...) {

#ifdef PRINT_DEBUG

    va_list args;

    /* args point to the first variable parameter */
    va_start(args, format);
    ef_print("[Flash] (%s:%ld) ", file, line);
    /* must use vprintf to print */
    rt_vsprintf(log_buf, format, args);
    ef_print("%s", log_buf);
    va_end(args);

#endif

}

/**
 * This function is print flash routine info.
 *
 * @param format output format
 * @param ... args
 */
void ef_log_info(const char *format, ...) {
    va_list args;

    /* args point to the first variable parameter */
    va_start(args, format);
    ef_print("[Flash] ");
    /* must use vprintf to print */
    rt_vsprintf(log_buf, format, args);
    ef_print("%s", log_buf);
    va_end(args);
}
/**
 * This function is print flash non-package info.
 *
 * @param format output format
 * @param ... args
 */
void ef_print(const char *format, ...) {
    va_list args;

    /* args point to the first variable parameter */
    va_start(args, format);
    /* must use vprintf to print */
    rt_vsprintf(log_buf, format, args);
    rt_kprintf("%s", log_buf);
    va_end(args);
}

xiaodingding, 2020-05-20:
#include <fal.h>

#include "at24cxx.h"

#define AT24CXX_PAGE_SIZE	64	

static uint8_t erase_buff[AT24CXX_PAGE_SIZE] = {0};

// extern sfud_flash sfud_norflash0;
static at24cxx_device_t at24cxx_dev;
static int init(void)
{
    at24cxx_dev = at24cxx_init("i2c1", 0);
	if(NULL == at24cxx_dev)
	{
        rt_kprintf("at24cxx_init for fal init failed \n");
	}else
    {
        rt_kprintf("at24cxx_init for fal init successful \n");
    }
    return 0;
}

static int read(long offset, uint8_t *buf, size_t size)
{
    // assert(sfud_norflash0.init_ok);
   // sfud_read(&sfud_norflash0, nor_flash0.addr + offset, size, buf);
    if(RT_EOK != at24cxx_page_read(at24cxx_dev, offset, buf, size))
    {
		rt_kprintf("at24cxx_page_read error \n");
        return -1;
    }
//	rt_kprintf("at24cxx_page_read offset:%d size:%d \n", offset, size);
    return size;
}

static int write(long offset, const uint8_t *buf, size_t size)
{
    // assert(sfud_norflash0.init_ok);
    // if (sfud_write(&sfud_norflash0, nor_flash0.addr + offset, size, buf) != SFUD_SUCCESS)
    // {
    //     return -1;
    // }
	
	if(size < AT24CXX_PAGE_SIZE)
	{
	
		if(RT_EOK != at24cxx_page_write(at24cxx_dev, offset, (uint8_t *)buf, size))
		{
			rt_kprintf("at24cxx_page_write error \n");
			return -1;
		}
		
	}else
	{
		if(RT_EOK != at24cxx_page_write(at24cxx_dev, offset, (uint8_t *)buf, size))
		{
			rt_kprintf("at24cxx_page_write error \n");
			return -1;
		}
	}
//	rt_kprintf("at24cxx_page_write offset:%d size:%d \n", offset, size);
    return size;
}

static int erase(long offset, size_t size)
{
#if 1	
    rt_err_t result = RT_EOK;
	const uint16_t earse_len = sizeof(erase_buff);
    rt_memset(&erase_buff[0], 0, earse_len);
	
	rt_kprintf("at24cxx earse write offset:%d ,size:%d \n", offset , size);
	
    if(size <=  earse_len )
    {
        if(RT_EOK != (result = at24cxx_page_write(at24cxx_dev, offset, erase_buff, size) ) )
        {
            rt_kprintf("at24cxx earse at24cxx_page_write to  earse is error result:%d \n", result);
            return -1;
        }
    }else
    {
        uint16_t  yushu = 0, i = 0;
        const uint16_t loop = size /  earse_len;

        for(i = 0; i < loop; i++)
        {
            if(RT_EOK != (result = at24cxx_page_write(at24cxx_dev, offset + (earse_len * i)  , erase_buff,  earse_len) ) )
            {
                rt_kprintf("at24cxx earse at24cxx_page_write to  earse is error result:%d \n", result);
                return -1;
            }
			rt_kprintf("at24cxx earse write addr:%d ,yushu:%d, result:%d \n", offset + (earse_len * i) , i, result);
        }

        yushu = size % earse_len;
		rt_kprintf("at24cxx earse  offset:%d ,size:%d ,yushu:%d ,loop:%d  \n", offset , size, yushu, loop);
		if(yushu > 0)
		{
			if(RT_EOK != (result = at24cxx_page_write(at24cxx_dev, offset + (earse_len *loop) , erase_buff, yushu) ) )
			{
				rt_kprintf("at24cxx earse at24cxx_page_write to  earse is error result:%d \n", result);
				return -1;
			}
			rt_kprintf("at24cxx last earse  addr:%d ,yushu:%d, result:%d \n", offset + (earse_len * loop) , yushu, result);
		}else
		{
			rt_kprintf("at24cxx earse at24cxx_page_write to earse size:%d \n", size);
		}

    }
	rt_kprintf("at24cxx earse at24cxx_page_write to earse size:%d offset:%d\n", size, offset);
    return size;
#else
	rt_kprintf("at24cxx earse is null function \n");
	return size;
#endif
}
const struct fal_flash_dev eeprom0 = { "eeprom0", 0, 16*1024, 64, {init, read, write, erase} };