armink/EasyFlash #104: The ENV (@0x004B409C) CRC32 check failed!

mingminglove, 2020-06-11:
Hi armin，
我们现在遇到一个非常棘手的问题，我们产品马上要量产了。但是最近客服测试产品过程不断的看门狗超时复位。我们现实调试发现在写flash过程中出现错误一旦出现这个错误就会复位。
[Flash]Error: The ENV (@0x004B409C) CRC32 check failed!
[Flash]Error: The ENV (@0x0053A080) CRC32 check failed!
[image]
[image]

armink, 2020-06-11:
先关闭看门狗，接上仿真器看下后面代码的运行情况吧

mingminglove, 2020-06-11:
Hi armink,
我关闭看门狗就不会出现复位的情况了。我测试之后发现了规律，就是我打开看门狗第一次写这个数据超时，然后复位，下次写的时候就会出现error,我把看门狗关掉这个error就不会出现了。看门狗打开：第一次写复位：Diagnostic positive response received: ID:1f40f424 payload: 02f621245 0 09cd0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 e80 0 0 11e 0 0 0 0 0 0 0 0 0 0 0 0
第二次写复位：Diagnostic positive response received: ID:1f40f424 payload: 02f621245 0 09cd0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 e80 0 0 11e 0 0 0 0 0 0 0 0 0 0 0 0
[Flash]Error: The ENV (@0x00541C4C) CRC32 check failed!
第三次关闭看门狗：Diagnostic positive response received: ID:1f40f424 payload: 02f621245 0 09cd0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 e80 0 0 11e 0 0 0 0 0 0 0 0 0 0 0 0
[Flash]Error: The ENV (@0x00541C4C) CRC32 check failed!不复位不死机
第三次关闭看门狗：
Diagnostic positive response received: ID:1f40f424 payload: 02f621245 0 09cd0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 e80 0 0 11e 0 0 0 0 0 0 0 0 0 0 0 0
complete log save success
数据存储成功

armink, 2020-06-11:
那是不是你的看门狗超时时间太短了，擦写 flash 时，会占用 CPU 的

mingminglove, 2020-06-11:
可能是这种原因，后面我们更改喂狗策略，现在没问题了。非常感谢您耐心的回答。