armink/EasyFlash #105: 环境变量key写入问题

mingminglove, 2020-06-29:
Hi armink,
我们产品市场长期工作之后问题显露出来。
问题1：调用API接口ef_set_env_blob写入大量的key这样easyflash_init初始化时间会增加，并且再次ef_set_env_blob时间也会相应的增加。我调用ef_del_env删除所有环境变量，但是初始化时间并没有减少，ef_set_env_blob时间也没变化还是很长。不知道怎么解决？
问题2：这个时间增加不是只和key的多少有关，而我反复调用ef_set_env_blob写同样的key值，key没有增加但是反复写入。easyflash_init初始化和ef_set_env_blob时间也会增加。不知道怎么解决？
问题3：当我调用ef_set_env_blob写入同一个key值磨损均衡，当这个ENV_AREA_SIZE区域写满时，是循环继续从是EVN首地址写吗？

armink, 2020-06-29:
你的数据量有多大？可以试试 legacy 模式？

[image]

armink, 2020-06-29:
用 legacy 时需要有充足的内存，200*2K = 400K，你先详细看下 README 里的介绍

mingminglove, 2020-06-30:
Hi armink, 
             感谢您耐心的回答，200*2K = 400K这个内存开销太大了，我用的M4内核最大内存200K现在应用层已经用了大部分，剩余给easyflash不足8K。如果不用legacy 模式，NG模式ef_set_env_blob多次写入同一个key会导致时间增加是不可避免无法解决的，对吗？

armink, 2020-06-30:
同一个 key 写入多次，内部的操作是先删除以前的，再插入新的。这里的删除操作，不一定是真正的删除，而只是做了脏标记，下次检测到这个脏 key ，就自动跳过了。所以时间长了以后，脏的越来越多，查询时间也就越来越长，也许手动再后台，不定期调用 GC 后，这种情况也会有缓解，你可以试试。

另外你现在配置的 ENV 分区的容量是多大？

mingminglove, 2020-06-30:
好的，我试试。我现在分配给ENV的容量是26M。

armink, 2020-06-30:
26MB？那确实太大了，空间给 200*2K*4 差不多就够了

mingminglove, 2020-06-30:
是太大，到时候我按实际的内存给他分配。是不是如果一个key写满ENV的容量，下一次再次写入key的时候就会从首地址覆盖脏标记内容。

armink, 2020-06-30:
KEY 写满了，会自动执行 GC 的，以前脏的 KEY ，都会自动被清理掉，所以 GC 后的执行速度又会变快了

mingminglove, 2020-06-30:
明白了，多谢。

HikerPan, 2020-07-03:
@armink 今天自己研究了easyflash的代码，我觉得easyflash存储数据大了以后，速度变慢，主要是因为每次都要从flash的首地址开始按照扇区查找并确定各个扇区的状态。
ef_cfg.h中，对norflash的最小擦除粒度是4K，由于还有32和64K的block，我是否可以将最小擦除粒度修改为64K，这样每次查找的次数就会减少16倍？
不知道我理解的对不对