armink/EasyFlash #107: 代码逻辑问题

shaowencai, 2020-08-14:
[image]
这块的代码感觉永远不会执行

i-jaffer, 2022-12-07:
刚开始我也是这样理解的，后来发现错了，这里代码应该没有问题

需要注意的是对index的理解，index是逻辑地址，不是物理地址

比如：flash的物理地址可能是0-100，index是可以支持 100 - 120读取的，这样理解你应该就能看懂了

zhangyanlei830, 2023-01-12:
我觉得应该根据这段代码一起看，看注释我是这样想的，不知道对不对。
`/**
 * Calculate flash physical address by log index.
 *
 * @param index log index
 *
 * @return flash physical address
 */
static uint32_t log_index2addr(size_t index) {
    size_t header_total_offset = 0;
    /* total include sector number */
    size_t sector_num = index / (EF_ERASE_MIN_SIZE - LOG_SECTOR_HEADER_SIZE) + 1;

    header_total_offset = sector_num * LOG_SECTOR_HEADER_SIZE;
    if (log_start_addr < log_end_addr) {
        return log_start_addr + index + header_total_offset;
    } else {
        if (log_start_addr + index + header_total_offset < log_area_start_addr + LOG_AREA_SIZE) {
            return log_start_addr + index + header_total_offset;
        } else {
            return (log_start_addr + index + header_total_offset) - LOG_AREA_SIZE; // 如果这样那么你说的这段代码确实不执行
            // return log_start_addr + index + header_total_offset;//这样就没问题了
        }
    }
}`