armink/EasyFlash #111: FLASH_TYPEPROGRAM_DOUBLEWORD是指颗粒度是64位吗？ STM32L552ZE是否支持？

augussong, 2020-10-08:


augussong, 2020-10-09:
D/NO_TAG          (../easyflash/src/ef_env.c ef_env_init:1821)ENV start address is 0x08010000, size is 4096 bytes.

I/NO_TAG          Warning: Sector header check failed. Format this sector (0x08010000).

I/NO_TAG          Warning: Sector header check failed. Format this sector (0x08010800).

I/NO_TAG          Warning: All sector header check failed. Set it to default.

I/NO_TAG          EasyFlash V4.1.99 is initialize success.

I/NO_TAG          You can get the latest version on https://github.com/armink/EasyFlash .

V/main            [10:08:12 pid:1008 tid:24] (E:/Projects/Aura/STM32L5_TEST/LPUART/Core/Src/main.c main:150)Reboot_Time is 1

V/main            [10:08:12 pid:1008 tid:24] (E:/Projects/Aura/STM32L5_TEST/LPUART/Core/Src/main.c main:154)Reboot_Time is 2

armink, 2020-10-09:
V4 版本的新模式不确定，你可以试试

但 V3 版本遗留的 legacy  模式应该支持，详见 ef_env_legacy.c

augussong, 2020-10-10:
V3 版本遗留的 legacy 模式是可以的，希望V4 或FlashDB 也可以支持

armink, 2020-10-10:
主要问题并不是 64 bit 写粒度导致无法支持，而是像 STM32L4 这种 64 bit 写粒度 Flash ， **没法支持逆序写入** ，所以不能用 V4 版本

closed 2020-10-12 by augussong