armink/EasyFlash #115: log_seq_read 读取指定 size 的数据与实际存储在 flash 上的数据偏差12个字节？

FJTBC, 2021-02-19:
原函数实现：
static EfErrCode log_seq_read(uint32_t addr, uint32_t *log, size_t size) {
    EfErrCode result = EF_NO_ERR;
    size_t read_size = 0, read_size_temp = 0;

    while (size) {
        /* move to sector data address */
        if ((addr + read_size) % EF_ERASE_MIN_SIZE == 0) {
            addr += LOG_SECTOR_HEADER_SIZE;
        }
        /* calculate current sector last data size */
        read_size_temp = EF_ERASE_MIN_SIZE - (addr % EF_ERASE_MIN_SIZE);
        if (size < read_size_temp) {
            read_size_temp = size;
        }
        result = ef_port_read(addr + read_size, log + read_size / 4, read_size_temp);
        if (result != EF_NO_ERR) {
            return result;
        }
        read_size += read_size_temp;
        size -= read_size_temp;
    }

    return result;
}

修改之后的函数实现：
static EfErrCode log_seq_read(uint32_t addr, uint32_t *log, size_t size) {
    EfErrCode result = EF_NO_ERR;
    size_t read_size = 0, read_size_temp = 0;
    uint32_t temp_addr = addr;

    while (size) {
        /* move to sector data address */
        if (temp_addr % EF_ERASE_MIN_SIZE == 0) {
            temp_addr += LOG_SECTOR_HEADER_SIZE;
        }
        /* calculate current sector last data size */
        read_size_temp = EF_ERASE_MIN_SIZE - (temp_addr % EF_ERASE_MIN_SIZE);
        if (size < read_size_temp) {
            read_size_temp = size;
        }
        result = ef_port_read(temp_addr, log + read_size / 4, read_size_temp);
        if (result != EF_NO_ERR) {
            return result;
        }
        read_size += read_size_temp;
        size -= read_size_temp;
        temp_addr += read_size_temp;
    }

    return result;
}

liuchao20080910, 2021-03-16:
> 
> 调用 log_seq_read 读取 8192 字节长度的数据，会少读取12个字节。
> 
> 原函数实现：
> 
> ```
> static EfErrCode log_seq_read(uint32_t addr, uint32_t *log, size_t size) {
>     EfErrCode result = EF_NO_ERR;
>     size_t read_size = 0, read_size_temp = 0;
> 
>     while (size) {
>         /* move to sector data address */
>         if ((addr + read_size) % EF_ERASE_MIN_SIZE == 0) {
>             addr += LOG_SECTOR_HEADER_SIZE;
>         }
>         /* calculate current sector last data size */
>         read_size_temp = EF_ERASE_MIN_SIZE - (addr % EF_ERASE_MIN_SIZE);
>         if (size < read_size_temp) {
>             read_size_temp = size;
>         }
>         result = ef_port_read(addr + read_size, log + read_size / 4, read_size_temp);
>         if (result != EF_NO_ERR) {
>             return result;
>         }
>         read_size += read_size_temp;
>         size -= read_size_temp;
>     }
> 
>     return result;
> }
> ```
> 
> 修改之后的函数实现：
> ```
> static EfErrCode log_seq_read(uint32_t addr, uint32_t *log, size_t size) {
>     EfErrCode result = EF_NO_ERR;
>     size_t read_size = 0, read_size_temp = 0;
>     uint32_t temp_addr = addr;
> 
>     while (size) {
>         /* move to sector data address */
>         if (temp_addr % EF_ERASE_MIN_SIZE == 0) {
>             temp_addr += LOG_SECTOR_HEADER_SIZE;
>         }
>         /* calculate current sector last data size */
>         read_size_temp = EF_ERASE_MIN_SIZE - (temp_addr % EF_ERASE_MIN_SIZE);
>         if (size < read_size_temp) {
>             read_size_temp = size;
>         }
>         result = ef_port_read(temp_addr, log + read_size / 4, read_size_temp);
>         if (result != EF_NO_ERR) {
>             return result;
>         }
>         read_size += read_size_temp;
>         size -= read_size_temp;
>         temp_addr += read_size_temp;
>     }
> 
>     return result;
> }
> ```
> 

原函数这句话read_size_temp = EF_ERASE_MIN_SIZE - (addr % EF_ERASE_MIN_SIZE);改成read_size_temp = EF_ERASE_MIN_SIZE - ((addr + read_size) % EF_ERASE_MIN_SIZE);也可以。

liuchao20080910, 2021-03-16:
原函数这句话read_size_temp = EF_ERASE_MIN_SIZE - (addr % EF_ERASE_MIN_SIZE);改成read_size_temp = EF_ERASE_MIN_SIZE - ((addr + read_size) % EF_ERASE_MIN_SIZE);也可以。