armink/EasyFlash #124: ef_env.c中“set_status”这个接口内部的status_table有问题？

zhoujie-jay, 2021-07-26:
对于
...
#if (EF_WRITE_GRAN == 1)
        byte_index = (status_index - 1) / 8;
        status_table[byte_index] &= ~(0x80 >> ((status_index - 1) % 8));
#else
...
中的0x80，如果编译器把0x80当成unsiged char 类型的移位结果和 当成char类型的移位结果是截然不同。
所以说你想表达的结果应该是status_table[byte_index] &= ~(（char）0x80 >> ((status_index - 1) % 8)); @armink

armink, 2021-07-26:
或者改为  0x80UL ？

zhoujie-jay, 2021-07-26:
不能改成0x80UL的。假如使用你说的，我要将ENV_PRE_DELETE状态变成ENV_DELETED，此时代码如下： result = write_status(old_env->addr.start, status_table, ENV_STATUS_NUM, ENV_PRE_DELETE);
result = write_status(old_env->addr.start, status_table, ENV_STATUS_NUM, ENV_DELETED);
此时将ENV_PRE_DELETE写到flash中（地址：0x460d4），写入的内容是0xDF(1101 1111)。接着状态变成ENV_DELETED，此时将ENV_DELETED写入到flash中（地址：0x460d4），写入的内容是0xEF(1110 1111)。所以如果要将0x460d4地址（0xDF）的内容改写成0xEF，有可能写入不成功，即0xEF的值写入到更新到flash失败。如果是（char）0x80 就不会存在该问题。
get_status 接口中的
#if (EF_WRITE_GRAN == 1)
        if ((status_table[status_num / 8] & ((char)0x80 >> (status_num % 8))) == 0x00) {
            break;
        }
#else /*  (EF_WRITE_GRAN == 8) ||  (EF_WRITE_GRAN == 32) ||  (EF_WRITE_GRAN == 64) */
0x80，是否也要做调整？

armink, 2021-07-26:
没看太懂你的解释呢