armink/EasyFlash #141: 请问Flash读写会不会有被中断打断的隐患

tovax, 2022-06-17:
当有RTOS的情况下，lock和unlock通过信号量实现，请问会不会存在“flash_write”的时候被中断打断的隐患？
```
/**
 * lock the ENV ram cache
 */
void ef_port_env_lock(void) {
    rt_sem_take(&env_cache_lock, RT_WAITING_FOREVER);
}

/**
 * unlock the ENV ram cache
 */
void ef_port_env_unlock(void) {
    rt_sem_release(&env_cache_lock);
}
```

armink, 2022-06-17:
如果你写 flash 时候中断是开的，肯定有这个可能

tovax, 2022-06-18:
> 

嗯，实际应用的时候一般都有中断，这里的信号量估计得用开关中断，或者进入临界区来保护

armink, 2022-06-18:
为什么呢？你要在中断里使用 EasyFlash ？

tovax, 2022-06-18:
> 

那倒不是，我的意思在普通的任务中使用EasyFlash，那么EasyFlash在写入数据的时候，会不会被串口接收中断、脉冲捕获中断之类的干扰了？不是在中断服务函数里使用EasyFlash。

tovax, 2022-06-18:
lock和unlock之间的代码会被别的中断打断吗

armink, 2022-06-18:
lock 目的就是为了锁住上下文，如果你中断中不使用 EasyFlash 那就没事的

tovax, 2022-06-18:
好的，谢谢你的回复和工作成果。准备在GD32+FreeRTOS项目中使用EasyFlash。

armink, 2022-06-18:
可以试试 RT-Thread + EasyFlash ，有现成的 demo https://github.com/armink/EasyFlash/tree/master/demo/env/stm32f4xx，还有软件包： https://packages.rt-thread.org/detail.html?package=EasyFlash

tovax, 2022-06-18:
好的